biologist79/ESPuino #1: RC522 very slow / unresponsive with this code

georg90, 2020-05-04:
Hi! First of all - great job! I was looking for adding FTP support to the Tonuino project when I found your repo, thanks for the great documentation.
I really like the way of handling audio on the esp rather than using the dfplayer.

When trying out your image, I noticed that the RFID reader is quit slow / doesn't pick up cards as easy as with sample code from the repo. 

I changed cables etc, but when I use a basic sketch to try out the chip, it works as supposed.

I had a quick look at your code and couldn't find a delay other than the 300ms in line 1467.

Since I am not too much into the library for the RC522, my fist guess is, that line https://github.com/biologist79/Tonuino-ESP32-I2S/blob/bfb8d81af2914f16de721057272625c5ad056082/src/main.cpp#L1471 will check for a new card. If I hold the same card, it will not trigger. However, if I try three different tags, all not known to the device, it will only pick up 1 of them instantly.

I will investigate further, maybe others have the same issue and we can exchange ideas here. Thanks!

biologist79, 2020-05-05:
Thanks for your feedback!

Did this coding-part in the very beginning of my project. When I remember right my goal was to eliminate issues with triggering "new card present" too many times at once. However, try to set it to 1 ms and see, if it fits your needs.

Anyway, I'm aware there's a fork that was especially made for changing rfid-behaviour in the way that music stops, when the card is removed (https://github.com/mcgreg1/Tonuino-ESP32-I2S/blob/master/src/main.cpp#L1573). As it seems he has reduced the "rfid-debounce" to 100 ms as well.

georg90, 2020-05-06:
It's better now. Thanks for the hint ;-) 100ms seems to help.

I will create a fork and upload my code at some point. Maybe its helpful (I added remote debugging and changed the button logic to my needs. and a few other things)

closed 2020-05-06 by georg90