biologist79/ESPuino #2: Feature: Add Audio Feedback

georg90, 2020-05-06:
Hi again.
I am currently trying to implement audio feedback. So e.g. when the box starts, next track etc. there is also a voice output.

Currently I see two options:
1. Working with the Queue system in place (drawback: When interrupting a queue (=playlist) we loose the state of where we where.
2. Have an independet implementation that handles audio feedback.

I like option 2, I tried option 1 and it breaks the playlist, when being used within a queue being played. For option 1 I am currently missing more information on how the MAX device and the I2S protocol is designed. Can we have a parallel connection ? Or how can we enable sound output while keeping te queue structure in place?

Any ideas welcome.

Thanks a lot.

biologist79, 2020-05-07:
One thing first: Is it really necessary to give an audio-feedback? Just asking because what I personally like more or less the most about my tonuino-project is the Neopixel So I ended up in showing some informations using leds. Don't want to say your way is wrong, just want to make sure you gave it a thought So for jumping to the first or last track there's already an animation available.

However, regarding your questions...
I2S: Just had a look at Espressif's specs and they say "ESP32 contains two I2S peripherals. These peripherals can be configured to input and output sample data via the I2S driver.". So at least in theory it should be possible to run two instances in parallel. You could try to use a second instance of the audio-library used but I don't know if there's enough global memory available to do that (was by far the biggest issue I had while developing my fork). If that happens (just as hint) the ESP will end up in a boot-loop. I just mention this as it took me a while to point out the problem. But to be honest I don't know if I2S technically supports such a 2:1-connection. Anyway, I2S seems to be very robust. If you play music for instance and disconnect an I2S-pin, the music is gone (and possibly you hear ugly noises) but after reconnecting music is immediately back. So I cannot answer it - just a few thoughts about it.

Changing playQ later is not easy
1) Guess you already recognized the big array (bit-field) starting at around line 130 that stores all properties of the playlist. I'd enrich this structure by uint8_t specialFileNumber and give it a 4 bit adress-space. As per default it's zero indicating that playing feedback-file is inactive. For states 1-7 I'd reuse what I called track-control (starting at around line 90). So 1 = stop; 2 = play... Additionaly, as you mentioned startup-sound is desired, maybe add an 8th state called startup or so. Next you need an array of char-arrays that contains the special file-names to play - call it maybe specialFiles[]. Make sure to store these char-arrays (strings) in PROGMEM to save memory. Inside there's something like "/mp3/specialFiles/pause.mp3". specialFiles[0] is null. Starting with specialFiles[1] you reuse track-control's numerical logic as this enables you to address this array later simply using the states of track-control.

2) For playAudio() re-use the existing track-control-structure. So for example if pause/play was hit, add a line in the section of pause-handling setting specialFileNumber = PAUSE. Same for next track and so on...

3) Introduce a global if/else-branch to distinguish between "normal play mode" und "special file mode". So if playProperties.specialFileNumber > 0 special-file-handling is called. When playling special file is over give control back to normal play mode by setting playProperties.specialFileNumber back to 0.

Should work this way in general but maybe in detail it's a bit more complicated

Things to mention:

a) For pause/play it's a bit complicated because you have to jump back into the old file at the right position. So you have to make sure to save the current playPosition every time pause/play is pressed. Currently playPosition is only saved in audiobook-mode.
b) Make sure vTaskDelay() is called if no track is played. If you don't do ESP will end up in a crash as the infinite loop will run that fast there's no time left for the ESP to manage other things (Wifi for example).

Hope that helps.

georg90, 2020-05-07:
First of all, thanks for the detailed description and thoughts. 
I, too, like the neopixels a lot! However, my enclosure currently doesn't feature LEDs other than for status (on/off). Therefore I'd like to enable audio output for the most basic tasks.

I followed you're approach and got it working (partially).

When the music playback is stopped, I can successfull play the audio commands using this function in `trackQueue`:

```
// Audio feedback handling
            #ifdef AUDIO_FEEDBACK_ENABLE
                if (playProperties.specialFileNumber > 0) {
                  playProperties.trackFinished = false;
                  loggerNl("Handling audio feedback", LOGLEVEL_NOTICE);
                  audio.connecttoSD(specialFiles[playProperties.specialFileNumber]); // this is the array with the path to the mp3
                  playProperties.specialFileNumber = 0; // setting back to default mode
                  audio.loop(); // starting the audio loop
                  vTaskDelay(portTICK_PERIOD_MS*10); // delaying the task to keep esp alive
                }
            #endif
```

When the box is playing, this is firing, too, but nothing is played (no audio output). My assumption is the following track (e.g. the next track) is overwriting the output of the audio. I though this is prevented by using `playProperties.trackFinished = false;` which is set to `true` when `eof_mp3` is called. But it seems this is still continuing and breaking the audio playback when a playlist is there.

biologist79, 2020-05-07:
I guess the problem is that you revert `playProperties.specialFileNumber` back to 0 immediately. Because in this case `audio.loop()` is only called exactly once. But this isn't sufficient as it has to be called on and on until track's end was reached. So make sure to allow this. 

Unfortunately the audio-lib used doesn't provide a method that allows checking if music is still playing. Would make things a bit easier but `eof_mp3` works too.

georg90, 2020-05-07:
Yes, I had this in mind and changed it already. But without success.

```
␛[0m(D) Kontroll-Kommando empfangen via Queue: 5
␛[0m(D) Handling audio feedback
␛[0m(D) info        : Reading file: 03_prev_track.mp3
␛[0m(D) info        : file has no mp3 tag, skip metadata
␛[0m(D) info        : stream ready
␛[0m(D) info        : syncword found at pos 0
␛[0m(D) info        : Channels=1
␛[0m(D) info        : SampleRate=44100
␛[0m(D) info        : BitsPerSample=16
␛[0m(D) info        : Bitrate=32000
␛[0m(D) info        : Reading file: Bobo Siebenschl04.mp3 // This is the next file (no eof_mp3 being called)
```

Using this in `eof_mp3`
```
void audio_eof_mp3(const char *info) {  //end of file
    snprintf(logBuf, sizeof(logBuf) / sizeof(logBuf[0]), "eof_mp3     : %s", info);
    loggerNl(logBuf, LOGLEVEL_INFO);
    #ifdef AUDIO_FEEDBACK_ENABLE
        if (playProperties.specialFileNumber > 0) {
            playProperties.specialFileNumber = 0;
        }
    #endif
    playProperties.trackFinished = true;
}
```

It seems the file is not played at all and being overwritten. `eof_mp3` is never called.

georg90, 2020-05-07:
Did some more try/error:

Moved the check up (above `trackQStatus`) and changed the logic to something like this:
```
        // Audio feedback handling
        #ifdef AUDIO_FEEDBACK_ENABLE
            if (playProperties.specialFileNumber > 0) {
              playProperties.trackFinished = false;
              loggerNl("Handling audio feedback", LOGLEVEL_NOTICE);
              audio.connecttoSD(specialFiles[playProperties.specialFileNumber]);
              while(!playProperties.trackFinished) {
                audio.loop();
              }
              vTaskDelay(portTICK_PERIOD_MS*300);
            }
        #endif
```
This works, but the played files are played with a little noice / cut at the beginning and end. Did you experience something like this with the MAX device? The files play fine on my PC. Maybe it's due to only being mono sound files? Longer files play fine..

biologist79, 2020-05-08:
Sounds good :-) Glad to hear my idea works.
Indeed I had such issues but wasn't sure if it's maybe a hardware-issue. To be honest I didn't do more investigastion on it as playing of short files was out of scope for me.

While developing the feature "play whole folder" I used some tiny mp3s in order to save time and doing a sort of stress test. In fact I took some files from the original tonuino-project that are used to spell numbers 1-4. So really short. Do not really remember if it was noisy but at least it was cut.

But yes, maybe mono is the problem, because usually (when I got it right) stereo-files are so to say multiplexed into a single channel. By having only one channel maybe this is an issue.

Regarding noises I recently had a private chat with somebody that reported ugly noises when changing volume fast via rotary encoder. In fact I also discovered this as a little bit of an issue but wasn't really "impressive". However, by reducing gain on the MAX for him it seems to be far better now. Per default it's 9dB and because volume @1 was too lood for me (primary usage is my son's sleeping room) I changed it very early down to 3 dB. So just as hint, don't know if that would things make better for you.

georg90, 2020-05-08:
Thanks again. Gain was something I was thinking about changing anyway. However, a quick test revealed it didn't help in this case.

For everyone ending up here from google etc.

I could "fix" it by adding some silence (50/100ms) in front and the back of the audio track. My files (<2 seconds) play fine now.

I consider this "done" now :)

closed 2020-05-08 by georg90

biologist79, 2020-05-09:
Good job!
Thanks a lot for your reply. Guess this is something I could consider to integrate into my repository as weel.