biologist79/ESPuino #4: Board

BluetriX, 2020-06-18:
Hi!

Ist geplant ein Board (PCB) für das Projekt zu machen? Ich habe gerade die Bauteile vor mir aber ich bin erst bei der Hälfte und es sieht schon wüst aus.

Ich könnte ein Board entwerfen und es zur Verfügung stellen. 

Könntest du dann den Entwurf checken?

biologist79, 2020-06-18:
Hi,

das ist ein recht komplexes Thema :-)
Auf https://discourse.voss.earth/ gibt es unter 3rd-Party (muss man 
allerdings registriert sein und hat dann, weil es eben 3rd-Party ist, 
auch nicht sofort Zugriff drauf) einen Thread 
(https://discourse.voss.earth/t/esp32-fork-tonuino-mit-i2s-und-smarthome-integration/4973), 
den ich im Zuge meiner Softwareentwicklung geöffnet habe. Dort gab es 
jedenfalls schon verschiedene Vorstöße von verschiedenen Leuten, die da 
was gestartet haben. Ich für meinen Teil habe bisher immer alles auf 
einer Streifenraster-Platine aufgelötet, weil ich für eine Entwicklung 
zu faul war. Insofern: Ja, da existiert was und grundsätzlich ist es 
auch wünschenswert, wenn man das etwas mehr in geordnete Bahnen lenkt, 
weil die ganze Theorie dahinter nicht jedermanns Sache ist. Tatsächlich 
hat mir ein User netterweise zwei Platinen zukommen lassen für einen 
ESP32s mit 38 Pins. Das erwähne ich deshalb, weil ich dabei auch gelernt 
habe, dass es unterschiedliche ESP32s gibt (meine haben nur 30 Pins).

Nun ist es aber so, dass verschiedene Leute unterschiedliche 
Vorstellungen haben, und insofern gibt es da unterschiedliche 
Bestrebungen und Schwerpunkte. Ich will in Kürze einen zweiten Tonuino 
bauen, der als Devel-Board einen Lolin32 (Vorgänger vom Lolin D32) 
benutzt. Der Grund dafür ist, dass der Lolin32 (und auch der D32) 
bessere Festspannungsregler haben (deutlich weniger Stromverbrauch im 
Deepsleep) und sich insofern besser für den Akkubetrieb eignen. 
Gleichzeitig haben sie auch einen Anschluss für einen LiPo-Akku und eine 
dafür integrierte Ladeelektronik. Und nicht zuletzt muss man zum Flashen 
keine Taste drücken, was insbesondere deswegen interessant ist, weil man 
zum Flashen dann den Tonuino nicht mehr öffnen muss.

Das Ganze bringt nur ein Problem mit sich: Im Akkubetrieb hat man keine 
5V mehr zur Verfügung. Für den RFID-Reader kein Problem, der braucht ja 
eh nur 3.3V. SD-Kartenleser muss man sich einen passenden besorgen, auch 
kein Ding. Der beschriebene DAC kommt auch mit 3.3V aus; passt 
ebenfalls. Nur den Neopixel betreibt man dann außerhalb der Spec, weil 
der eigentlich für 5V gedacht ist. Aber ich habs mal ausprobiert: Der 
läuft auch mit 3.3V. Heißt aber natürlich, dass man die 
Transistorschaltung auf 3.3V anpassen muss, was auf jeden Fall andere 
Bauteile nötig macht. Ich habe vor einer Weile mal zwei Mosfets 
bestellt: NDP6020P und IRF530NPBF. Allerdings habe ich noch nicht 
ausprobiert, ob das auch wirklich funktioniert, damit 3.3V zu schalten.

Ansonsten gibts noch Leute, die arbeiten mit einer Powerbank - versorgen 
also weiterhin mit 5V. Hier hat man jedoch das Problem, dass diese im 
Deepsleep alles abschaltet und man dann schauen muss, wie man sie wieder 
aufweckt. Aber auch hierfür gibt es Lösungen (mit denen ich mich im 
Detail nicht befasst habe), aber ich fand die Variante mit LiPo direkt 
an den Lolin 32 "sexy".

Insofern entwickelt sich das Ganze gerade in verschiedene Richtungen und 
ich kann dir leider nix Fertiges präsentieren. Die Lolin32-Variante ist 
so gesehen schon bei Entwicklung etwas obsolet, da der Lolin32 halt 
durch den Lolin D32 abgelöst wurde (was ich bei Bestellung aber nicht 
wusste und jetzt habe ich ihn halt hier). Aber im Endeffekt ändert sich 
dadurch nur die Pin-Belegung (nehme ich mal an).

Vielleicht schaust mal in o.g. Forum vorbei im genannten Thread. Da 
findest auf jeden Fall mehrere Leute, die aktuell dieses Bestreben 
haben.

biologist79, 2020-06-18:
Achso, wenn du was entwickelst und mir zukommen lässt, dann kann ich mir 
das natürlich auch gerne mal anschauen.  Aber vielleicht musst du dir 
die Arbeit auch gar nicht machen. Denn so, wie ich das Projekt aktuell 
beschrieben habe (ESP32s, jedoch 38 Pins), gibt es bereits mindestens 
eine fertige Platine.

jobe451, 2020-10-04:
Hallo 
Leider scheint der obige Link zum Discourse-Thema nicht mehr zu funktionieren. 

Also habe ich habe den Versuch unternommen ein Board zu erstellen:
https://easyeda.com/jobe451/tonuino-esp32-i2s

- ESP 32 DevKitC
https://de.aliexpress.com/item/32950878456.html?spm=a2g0s.9042311.0.0.27424c4dD4RElf
- SD-Card-Reader (Regular und Micro)
https://de.aliexpress.com/item/32523546123.html?spm=a2g0s.9042311.0.0.27424c4dTV2BOP
- Lautsprecher
https://de.aliexpress.com/item/4000103165668.html?spm=a2g0s.9042311.0.0.27424c4dKKr5zp
- Verstärker
https://de.aliexpress.com/item/32999952454.html?spm=a2g0s.9042311.0.0.5e2d4c4de9Tqyo
- RFC Reader
https://de.aliexpress.com/item/32647697060.html?spm=a2g0s.9042311.0.0.5e2d4c4de9Tqyo
- Neopixel
https://de.aliexpress.com/item/32835427711.html?spm=a2g0s.9042311.0.0.5e2d4c4de9Tqyo
- Rotary Encoder: 
https://de.aliexpress.com/item/33041814942.html?spm=a2g0s.9042311.0.0.5e2d4c4de9Tqyo

Anmerkungen: 
- Ich habe versucht einen Verpolungsschutz einzubauen und dazu auch einen IRF9520 vorgesehen. Ob das so klappt weiss ich nicht, theoretisch, ja - bin da aber ein ziemlicher Amateur.
- Ich plane mehrere Boards zu bauen und dazu reguläre SD-Karten oder Micro-SD-Karten zu nutzen. Entsprechend hat es Anschlussstellen für beide Varianten.
- RFID-Karten, Buttons, Neopixel, usw sehe ich vor mit Dupont-Steckern zu verbinden.

Ein Feedback dazu würde ich sehr schätzen.

biologist79, 2020-10-04:
Hi,

der Link funktioniert schon. Das Problem ist nur, dass man im Forum angemeldet sein muss und selbst wenn man das ist, kriegt man das auch dann nicht gleich zu sehen. Grund hierfür ist, dass mein Entwicklungszweig von dem, was die Urheber des Projektes entwickelt haben, komplett abweicht (teilweise funktionell, aber vor allem von der Hardware und insbesondere der Codebasis). Insofern gilt mein Projekt dort als 3rd-party. Und da sich das Originalprojekt insbesondere auch an Anfänger richtet, wird 3rd-party am Anfang erstmal ausgeblendet, um diese nicht zu verunsichern. Ist ein bisschen schade, aber gut, ich bin dort mit meinem Projekt gewissermaßen nur zu Gast und gebe die Regeln nicht vor.

Zum Rest: Ich schaue mir das heute Abend mal genauer an und gebe dann Feedback. Ich wollte ja selbst schon lange eine solche Platine fertig haben, aber irgendwie ist in den Sommermonaten die große Faulheit eingekehrt :-) Ich habe zwar angefangen einen Plan zu zeichnen, aber als es dann zu konkreten Überlegungen wie Kopfhöreranschluss kam, habe ich es dann erstmal aufgeschoben.

biologist79, 2020-10-04:
@jobe451 Sieht ok aus würde ich sagen. Was ich an deiner Stelle noch einbauen würde, ist eine Reduzierung des Gains (https://cdn-shop.adafruit.com/product-files/3006/MAX98357A-MAX98357B.pdf Seite 28). Weil ich habe die Erfahrung gemacht, dass sonst die 21 Lautstärkestufen im unteren Bereich etwas grob gestuft sind. Oder anders gesagt: Ich konnte manchmal nicht so leise abspielen, wie ich es gerne getan hätte.
Zum Verpolschutz kann ich wenig sagen. Außer, dass ich eine Seite gefunden habe, wo es so beschrieben stand, wie du es gemacht hast.

biologist79, 2020-11-12:
Sodele, also mein eigenes Board ist jetzt endlich mal auf der Zielgeraden. Also zumindest geplant habe ich es mal und das Layout erstellt. Werde es dann in Kürze bei jclpcb bestellen. Es ist geplant für den Lolin32 mit einer zweistufigen Mosfet-Schaltung, die 3.3V für MAX98357a, Neopixel, RFID-Reader und SD-Reader schaltet. Dazu ein 6Pin-Konnektor zu einer Kopfhörerplatine, die optional drangesteckt werden kann. Wird dort ein Kopfhörer eingesteckt, wo wird der MAX98357a automatisch ausgeschaltet und umgekehrt. Die Kopfhörerplatine ist in SMD-Technik aufgebaut und stammt nicht von mir, aber ich werde den Urheber mal fragen, inwiefern ich diese hier publizieren oder verlinken darf.

tuffelh, 2020-12-03:
Gibt es dazu schon neues?
Wäre halt ein tolles "Referenzboard" :)

biologist79, 2020-12-03:
Nee, leider nicht. Ich warte auch schon auf die Lieferung von JLCPCB.
Würde es halt gerne mal getestet habe, bevor ich das hier publiziere. Wobei ich zugeben muss, dass ich, wenn ich vorher gewusst hätte, dass ich die Akkuspannung überwache, gleich den Lolin D32 (anstelle Lolin32) genommen hätte. Weil der hat den dafür notwendigen Spannungsteiler gleich mit drauf. Aber gut, jetzt habe ich die Lolin32 hier und nutze sie auch.

Gibt jetzt inzwischen verschiedene "Lager" an Nutzern:

- "Standard" ESP32-S: Die funktionieren, aber haben keine Ladeelektronik an Bord und brauchen auch im Standby durch die verbauten AMS1117-LDOs für Batteriebetrieb vergleichsweise viel Strom. Wie ich gelernt habe, gibt's diese Platinen auch in unterschiedlichen Formaten (glaube 32 vs 38 Pins). Nachteil an diesen ist: will man sie flashen, so muss man eine Taste auf dem Develboard drücken. Ist halt im eingebauten Zustand als Tonuino etwas doof. Das ist bei den Lolins besser.
- Lolin32: der Vorgänger von Lolin D32. Unterscheidet sich meines Wissens nur dadurch, dass man hier auch die Akkuspannung integriert messen kann und man die Spannung des Akkus über vbat auch direkt abgreifen kann. Den Lolin32 propagiere ich quasi.
- Lolin D32 pro: Besitzt zum D32 u.a. zusätzlich noch einen internen SD-Slot. Besitzt zusätzlich PSRAM, da als ESP32-Chip nicht der WROOM sondern WROVER verbaut ist. Habe mir gestern mal einen bestellt und will einfach mal schauen, ob ich mit dem PSRAM noch was Sinnvolles anfangen kann.
- ESP32-A1S-Audiokit: Eine fertige Platine, bei der man jedoch ein bisschen was löten muss. Hier ist der DAC bereits integriert in den ESP32. Habe diese Platine hier, da sie mir netterweise von einem User zur Verfügung gestellt wurde. Einen Proof-of-concept habe ich hier ja hochgeladen und ich werde das demnächst auch auf die aktuelle Masterentwicklung "heben". Aber für mich ist das Board aufgrund der wenigen GPIOs und des vergleichsweise hohen Standby-Strom nicht so interessant.

Kurzum: Bis auf Weiteres propagiere ich erstmal den Lolin32 und wenn es funktioniert, dann lade ich die Kicad- und Gerberfiles hier in Kürze hoch 👍

tuffelh, 2020-12-03:
Danke dir für das schnelle Feedback. Mir kommt es entgegen, dass du den Lolin32 genommen hast, da ich davon noch einige herumliegen habe.
Wenn du testen konntest werde ich mir ein paar der Boards bestellen, brauche gleich zwei für unsere Zwillinge...und ein paar als Reserve für Geschenke :-D
Welche Akkus planst du denn zu nutzen? Bzw, hast du schon eine BOM?

biologist79, 2020-12-03:
Was ist denn BOM? :-)
Wie auch immer: Via eBay habe ich damals einen von Eremit gekauft (der schreibt auch im Tonuino-Forum mit). Gibt's aber auch direkt bestellbar: https://www.eremit.de/c/3-7v-lipo-akkus/standart-lipo?page=2&sort=position-asc
Da habe ich jedenfalls einen hier zum Testen mit 2600 mAh. Ich kann es dir jetzt nicht genau quantifizieren, wie lange man damit hören kann, aber auf meinen USB-Multimeter zieht die Schaltung vom Netzteil mit aktiviertem WLAN im Betrieb so 160 mA etwa mit 24 Neopixel-LEDs, die auf 16/256 gedimmt sind und die ja auch selten alle zusammen leuchten. Klar, hörst lauter, ist es mehr. Also sagen wir vielleicht mal 200 mA. Also 10h gehen da sicher. Man kann natürlich auch einen größeren Akku nehmen, aber ich würde das mit den Lipo-Akkus in Kinderhand jetzt auch nicht übertreiben (meine persönliche Meinung). 2,x Ah waren für mich (gefühlt?) ein guter Kompromiss, obgleich ich mir nicht anmaße, ein Lipo-Experte zu sein :-)

tuffelh, 2020-12-03:
BOM== Bill of Material :-D
Also eine Liste der verbauten Komponenten. Dann könnte ich schauen was ich noch da habe und die fehlenden Sachen schon mal bei Ali bestellen.
Bei 10h Laufzeit bin ich voll bei dir, das sollte dicke reichen, sonst wird die Box auch zu schwer.

biologist79, 2020-12-04:
Auf der Hauptseite hier im Repository habe ich exemplarisch ja einiges beschrieben. Die Taster sind zB auf Aliexpress verlinkt. Den MAX98357a kann man von Adafruit kaufen, es gibt jedoch auch Clones, die genauso gut funktionieren. Einen solchen habe ich bei Ebay verlinkt, aber gibts auch bei Ali. Dann halt die beiden Mosfets IRF530NPBF und NDP6020P. RFID-Board ist hier in .de so günstig, ist eigentlich egal, wo man das bestellt. Neopixel-Ring kriegt man bei Ali erheblich günstiger. Habe hier in .de bestellt, aber eben mal geschaut: https://de.aliexpress.com/item/32673883645.html. SD-Reader würde ich für ein 3.3V-Setup einen ohne Spannungswandler nehmen, aber die muss man bisschen suchen. Habe ich zu eBay aber verlinkt. Drehencoder zB sowas hier: https://de.aliexpress.com/item/1005001506374915.html. Den Lautsprecher habe ich in Deutschland gekauft. Ja ansonsten wird es auf meiner Platine Konnektoren (JST-PH 2.0) statt Jumperwires geben. Dafür braucht's 2er, 3er und einen 5er. Sowas zB: https://de.aliexpress.com/item/4000091077742.html

biologist79, 2020-12-04:
Achso, JST-PH:
4x 2fach (Buttons, Batterieanschlus)
3x 3fach (Neopixel + RFID; den habe ich geteilt, um die 3er wiederzuverwenden)
1x 5fach (Drehencoder)

tuffelh, 2020-12-04:
Da fehlen mir ja nur die mosfets :)
Gleich Mal bestellt. Danke! Dann warte ich gespannt auf die Platine

biologist79, 2020-12-04:
[image]
[image]

Oben rechts ist ein Anschluss zur optionalen Kopfhörer-Platine. Diese habe ich hier ja bereits hochgeladen. MAX und SD werden einfach eingesteckt. Hab gerade vorhin gesehen, dass die beiden RFID-Stecker in unterschiedlicher Orientierung hier drauf sind. Aber das kannst du dir ja dann selbst aussuchen, wie rum der Stecker montiert ist :-)

biologist79, 2020-12-04:
Sodele, PCBs kamen eben. Heute Abend wird gelötet und getestet 👍

biologist79, 2020-12-05:
Ich habe irgendwie noch ein Problem mit dem RFID-Reader, das ich mir nicht erklären kann. Da muss ich nochmal schauen. Ansonsten läuft die Platine, nachdem ich den Widerstand am Gate des N-Channels-Mosfets von 4.7 auf 1k reduziert habe. Mir völlig unklar, warum das auf dem Breadboard immer funktioniert hat und eingelötet nicht.

biologist79, 2020-12-06:
So. Hab's hochgeladen. Ich werde demnächst auch nochmal einen PCB mit Lolin D32 machen. Da kann man sich den Spannungsteiler für die Messung der Batteriespannung sparen. Aber das wird ein paar Wochen dauern. Abgesehen von geringfügig mehr Arbeit funktioniert die Lolin32-Lösung aber genauso gut.

tuffelh, 2020-12-07:
Vielen Dank dafür! Gleich bestellt... Dann wird die neue Box noch zu Weihnachten fertig :-D

BluetriX, 2020-12-07:
Vielen Dank für das Bord - wenn ich die Gerberfiles (zip) zu jlcpcb hochlade gibt es keine Vorschau und er möchte die Bordmaße haben. Ich vermute ich mache einfach was falsch. Hast du einen Tipp?

biologist79, 2020-12-07:
Lässt einfach so stehen, wie es ist und klickst auf "Quote now". Danach ziehst du das ZIP-File rein und dann wird das selbst korrigiert.

BluetriX, 2020-12-07:
Ok - hab es nun nochmal genauso gemacht wie vorhin und nun kommt eine Vorschau etc, Vermutlich ein Schluckauf bei JLCPCB vorhin. Danke

biologist79, 2020-12-07:
Ich habe vorhin die Teileliste nochmal aktualisiert. Schaut, dass ihr auch den dort genannten uSD-Leser verwendet, damit auch Pin-Kompatibilität gewährleistet ist. Ich mache die Tage nochmal 2-3 Bilder von einer fertig gelöteten Platine und lade sie hoch.

erniberni, 2020-12-10:
hast du eventuell 1 Platine übrig? Würde meine Erfahrunge natürlich einbringen.
Ich habe heute alles als Kabelverhau aufgebaut, es funktioniert, aber ich denke mit Schrecken an den Einbau in ein Gehäuse

biologist79, 2020-12-10:
Also eine habe ich tatsächlich noch übrig. Schreib mir mal ne Mail an tonuino@ts-cs.de.

biologist79, 2020-12-10:
Eine Sache ist mir im Nachgang jetzt noch eingefallen: Und zwar wäre es ganz nett, wenn man einen Reset-Schalter hätte. Hintergrund ist: Betreibt man den Tonuino nicht im Batteriebetrieb, so kann man im Zweifelsfalle einfach den Stecker rausziehen. Mit Batterie geht das halt nicht. Ich habe mir jetzt für meine eigenen Tonuinos überlegt, dass ich einen Mikroschalter unten oder hinten am Gehäuse platziere und diesen an den Pins EN + GND anschließe. Drückt man das, so startet der ESP neu. Grundsätzlich wäre es natürlich schöner, diese beiden Pins auch auf der Platine zu haben. Ich lade nochmal demnächst eine rev2 hoch, bei der ich das mache. Ist jetzt für die rev1 aber auch kein Beinbruch: Lötet man die Drähte halt an EN + GND mit dran auf der Platine; so mache ich das bei mir jetzt auch.

biologist79, 2020-12-10:
Ich mache das hier jetzt mal zu. Wenn's dazu weitere Anmerkungen/Fragen gibt: Discussions hier habe ich vorhin freigeschaltet.

closed 2020-12-10 by biologist79

biologist79, 2020-12-11:
Jungs, ich habe eben eine aktualisierte Version des Lolin32-PCBs hochgeladen. Darin habe ich die eine JST-Buchse vom RFID-Reader rumgedreht, einen Reset-Pinheader hinzugefügt und einen Kondensator. Die Idee mit dem Reset kam mir erst die Tage. Und zwar wenn halt alles eingebaut ist in ein Gehäuse und man den Tonuino im Batteriebetrieb laufen hat, hat man keine Möglichkeit mehr, ihn neu zu starten. Einen Reset kann man erreichen, indem man GND auf EN schaltet. Genau die beiden Pins führe ich auch raus. Dort kann man dann zB einen Mikroschalter anschließen, den man halbwegs versteckt am Gehäuse platzieren kann. Ihr kennt das ja bestimmt von Geräten, die man kaufen kann. Da gibt's dann manchmal auch so kleine Löcher, wo man mit ner Stecknadel oder so rein muss. Also wenn ihr für sowas auch Verwendung habt, dann lötet euch halt einfach an EN und GND einen Draht mit dran und führt es zum Schalter. So mache ich das mit meinen bereits bestückten Platinen jetzt auch. 
Den 10 uF-Kondensator habe ich eingefügt, um die Spannungsmessung am Akku ein bisschen zu dämpfen. Habe ich mir beim Platinenlayout von Wemos beim Lolin D32 abgeschaut und es bringt tatsächlich auch was. Also muss man nicht machen, aber es dämpft auf jeden Fall die Schwankungsbreite der Wert. Habe den Kondensator bei mir jetzt einfach an der Unterseite der Platine mit angelötet; fertig.