biologist79/ESPuino #5: Kein Issue, danke!

mariolukas, 2020-10-11:
Ich wollte mich auf dem Weg kurz für die Arbeit bedanken die du hier reingesteckt hast. Ich konnte bisher eine von zwei Tonuinos auf deine ESP32 Lösung problemlos umbauen.

Um mir den Verkabelungs-Overhead so gering wie möglich zu halten habe ich mir auch ein Board in KiCad entworfen, nichts besonderes es entspricht so ca. dem Board was auch in #4 vorgestellt wurde. Falls interesse besteht kann ich es bei GitHub hochladen.

biologist79, 2020-10-11:
Hallo, vielen Dank!
Ja kannst du gerne machen. Ich werde öfter mal nach sowas gefragt, aber diesbzgl. bisher leider nichts vorzuweisen. Würde es dann vielleicht einfach in mein Repository übernehmen und aber natürlich auf dich verweisen. Ehre, wem Ehre gebührt :-)

mariolukas, 2020-10-28:
Ich bin leider noch nicht dazu gekommen. Es habe im Board noch einen kleinen Fehler gefunden, den ich zwar über eine Brücke lösen konnte, welcher aber im Layout besser raus sollte bevor ich es ins Repo packe.

biologist79, 2020-12-01:
Warte gerade auf PCBs von JLCPCB für den Lolin32. Wenn das alles passt, werde ich die Kicad/Gerber-Files hier in Kürze hochladen. Müssten eigentlich die Tage da sein die PCBs und dann weiß ich mehr.

mariolukas, 2020-12-06:
I think this can also be closed

closed 2020-12-06 by mariolukas