biologist79/ESPuino #6: Files (mp3) are not playing from the beginning.

mariolukas, 2020-10-27:
After a couple of weeks playing with the device i figured out that files don't play from the beginning. This behaviour does not match all my mp3 files. I tried several play modes. 

Anybody who can confirm a similar issue?

I am coming to the conclusion that this might be a mp3 encoding issue. I am using the suggested hardware components. 
Can some people please share some encoding information about the used mp3 files like sample rate etc. That might help debugging the issue.

biologist79, 2020-10-27:
Did you have a look at the debugging-commands at the serial interface? Just asking because it should show some information about at which position the file starts to play. Keep in mind when using audiobook-mode, the last play-position is saved when a new track starts or when you press pause. What you could do else is to have a look at backup.txt. All RFID-assigments/actions are saved there in a template like:
<file/folder>#<startPlayPositionInBytes>#<playmode>#<trackNumberToStartWith>
If <startPlayPositionInBytes> != 0, it won't start there. If in doubts: re-assign/learn RFID-tag to the action indended and check if it's working as expected then.

mariolukas, 2020-10-28:
> Did you have a look at the debugging-commands at the serial interface? Just asking because it should show some information about at which position the file starts to play.

The backup file was ok. Anyway i removed it for having a clean state. I have not selected audiobook-mode now. With a connected serial output i am getting thee following information.

A file that works:
```
info        : End of file test.MP3
eof_mp3     : test.MP3
info        : buffers freed, free Heap: 121696 bytes
info        : Reading file: 002.MP3
info        : MP3Decoder has been initialized, free Heap: 98052 bytes
info        : ID3 version=4
info        : ID3 framesSize=1176
info        : ID3 normal frames
```

A file that dit not work:
'/musik/test/01.MP3' wird abgespielt (2 von 12)
```
info        : stream ready
info        : syncword found at pos 117
info        : syncword found at pos 0
info        : MP3 decode error -2 : MAINDATA_UNDERFLOW
info        : syncword found at pos 0
info        : MP3 decode error -2 : MAINDATA_UNDERFLOW
info        : syncword found at pos 0
info        : Channels=2
info        : SampleRate=44100
info        : BitsPerSample=16
info        : BitRate=128000
```

`MP3 decode error -2 : MAINDATA_UNDERFLOW` shows that maybe something with my files is wrong. I am also wondering why the first sync word is found at pos 117.

biologist79, 2020-10-28:
Could you please this (short) sketch? https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/I2Saudio.ino
Try to play the corresponding file (needs to be hard-coded within the sketch). 
However, it's the lib I'm using to decode mp3s.

mariolukas, 2020-10-28:
I solved it. The problem was caused by id3 tags. Especially special chars seem to be a problem. In my case german umlauts (like Hörspiel) caused the issue. Due this issue might be a "hidden" one it should be mentioned in the Readme.md.

closed 2020-10-28 by mariolukas

biologist79, 2020-10-28:
Hmm, you're right, I should mention this. Indeed I thought I already did, but seems I only did @Tonuino-Forum. But I only knew Umlaute in filenames could lead to problems. Wasn't aware that id3-tag could be critical.
In fact I recently had a problem on my son's Tonuino with mp3s but didn't spend time on it yet. Maybe it's also due to id3-tag. Will have a look.
However, thanks for reporting!

mariolukas, 2020-10-28:
I just created a small bash script for normalising the file names and removing id3 tags. But be careful you should use it on a copy of your original files. This script changes the files and does not create a new copy of it.
It uses find and also depends on [id3convert](http://manpages.ubuntu.com/manpages/bionic/man1/id3convert.1.html) and rename. I think rename is installed by default on most linux distributions. I am using  macos and needed to install the tools with homebrew.

However it is a quick and dirty hack. Maybe it helps other people to prevent a manual renaming procedure.

```
#!/bin/bash

DIR=$1

find $DIR -depth -execdir rename 's/Ä/Ae/g;s/Ö/Oe/g;s/Ü/Ue/g;s/ß/ss/g' "{}" \;
find $DIR -depth -execdir rename 's/ä/ae/g;s/ö/oe/g;s/ü/ue/g;s/\"//g;s/\?//g' "{}" \;
find $DIR -depth -execdir rename "s/\'//g;s/\`//g;s/\'//g;s/\&/und/g;s/\ /_/g;y/A-Z/a-z/" "{}" \;

for file in $DIR/*; do
   #id3info $file;
   id3convert -s $file;
   #id3info $file;
done
```