biologist79/ESPuino #7: RC522 via second I2C instance

kkloesener, 2020-11-02:
Hey!
I want to thank you for the great project so far.

I have the ESP32-A1S Audio-Development-Kit here and i want to ask if its not possible to define a second I2C instance to use for the RC522. Personally i dont want to use the neopixel and rotary encoder. There would be enough GPIO for this and some keys connect (pause/play, next, stop)
Via this I2C there would also be an optional display possible.

This would be a good solution for everyone not willing to solder.

biologist79, 2020-11-02:
Hi,

AFAIK ESP32 is able to handle two i2c-instances. With having one already "fixed" it's necessary to declare a 2nd instance by assigning custom-GPIOs to it. Have to admit didn't test it yet but shouldn't be a big thing. Will give it a try.
But: if you don't want to use a rotary encoder what's your plan to change the volume? :-)

kkloesener, 2020-11-03:
Good Point.
as i wanted to put this in a child-friendly box, my first thought was to leave it at one level - or change via Web.
But when having a i2c I think about capacitive-touches like here: https://learn.adafruit.com/metal-inlay-capacitive-touch-buttons/overview
This way there are nearly no limits and even a "rotary" would be possible.

kkloesener, 2020-11-15:
After some Research i had to buy new RFID-Reader because the "standard" isn't able to communicate via I2C or UART without resoldering the chip :( 
i will start testing on this when everything arrives!

kkloesener, 2020-12-27:
Its complete implemented with
#40

closed 2020-12-27 by kkloesener

joewa, 2021-02-23:
Hey kkloesener!
> Its complete implemented with
> #40

Great! But can you please provide some more details how you got it the RC522 running via I2C with ESP32-A1S?
- Which RC522-board did you end up using with I2C?
- If this (https://www.az-delivery.de/products/rfid-set) is the "standard" you mean, it is not working for me too.
- Finally, what is the wiring for the board?

Here is a snippet of my settings.h:
```
//################## select SD card mode #############################
#define SD_MMC_1BIT_MODE              // run SD card in SD-MMC 1Bit mode
#define SINGLE_SPI_ENABLE             // If only one SPI-instance should be used instead of two (not yet working!) (Works on ESP32-A1S with RFID via I2C)

//################## select RFID reader ##############################
//#define RFID_READER_TYPE_MFRC522_SPI    // use MFRC522 via SPI
#define RFID_READER_TYPE_MFRC522_I2C  // use MFRC522
//#define RFID_READER_TYPE_PN5180       // use PN5180
```

Cheers,
Jörg

kkloesener, 2021-03-16:
Sorry for the delay.

I've used a board like [this](https://www.ebay.de/itm/13-56-MHz-RFID-Module-Card-Reader-Writer-Interface-I2C-IC-Sensor-Form-RC522-Card/172463236773?hash=item28279c26a5:g:Pn0AAOSwnHZYZBgq)
The Board you've mentioned is the "standard". You can use the board, but you need to resolder the Chip or drill a hole through the pcb (https://forum.arduino.cc/index.php?topic=442750.0). I was not successfully with this and ended up with two pcb with each a hole in the Chip :(

I need to check the exact wiring and will post it later

biologist79, 2021-03-16:
@kkloesener In my forum we've a discussing ongoing regarding the board used: https://forum.espuino.de/t/esp32-audio-kit-esp32-a1s/106
As you've been the one to finally integrate my esp32-a1s-"branch" into my main-branch: maybe you'd like to join us :-)