biologist79/ESPuino #9: esp32-audio-kit can't mount SD while MFRC is connected and MFRC is not working

eazy, 2020-11-12:
Hi,

thanks for your work getting tonuino running on ESP32-Audio-kit. I followed your instructions but it won't work with the ESP32-Audio-Kit.
The strange thing is, if I put all together with a normal ESP32-dev board everything works fine.

The Problem:
If I start the ESP-Audio-kit while everything is connected, the mount of the SD-Card fails and keeps trying in the loop. If I disconnect the SDA Wire from of the MFRC while the moutloop is running the loop is ending with a mounted SD-Card but RFID doesn't work. The interessting thing is that I don't get a message for the missing MFRC on boot.

If I'm booting the esp32-audio-kit with disconnected SDA wire of the MFRC the SD-Card can be monted on the first try and I get the message "WARNING: Communication failure, is the MFRC522 properly connected?".

In both cases RFID won't work.

What have I done:
I copied the whole content of the Folder "Hardware-Plaforms/ESP32-A1S-Audiokit" to the root folder of the Project and replaced all existing files. In the platformio.ini I removed the rfid lib "https://github.com/biologist79/rfid.git" and added "yveaux/AC101@^0.0.1".

All the rest is done like you discribed, I removed all the resistor for the Keys and solderd miso of the MFRC522 to Pin 7 of the SD-Card reader gpio 2.

Jumpers are set to off/on/on/off/on

Have you done anything else ? because its running fine with a normal esp32 dev board. Or maybe my board is broken!?

thanks

biologist79, 2020-11-12:
Just to make sure: you've been using RFID-Lib located here: https://github.com/biologist79/Tonuino-ESP32-I2S/tree/master/Hardware-Plaforms/ESP32-A1S-Audiokit/lib/MFRC522 ? I did it a bit dirty in platformio: Lib named in platformio.ini was fetched automatically. However, I overwrote it with that one from the link and deactivated this Lib-dep in platformio.ini subsequently.
And did you solder to the correct pin? #7 is, coming from the direction of the JTAG-GPIOs, the 3rd pin.

Have to say sorry that this part of the project is currently not fully integrated into my master-development-branch. There's still an pull-request ongoing but I'll have to test a bit first. Currently it's more proof-of-concept and to be honest I didn't expect people in being immediately interested in it 👍

eazy, 2020-11-12:
wow, that was a fast answer ;-)

Yes, I use the lib https://github.com/biologist79/Tonuino-ESP32-I2S/tree/master/Hardware-Plaforms/ESP32-A1S-Audiokit/lib/MFRC522 with all the other MFRC libs I'm not able to compile the code. Otherwise the code won't work with the normal esp32-dev board.

And the soldering is also correct, I doublechecked all the wiring and connectivity. Thats why I asked for maybe some other change on the board.

biologist79, 2020-11-13:
Hmm, maybe try this "isolated" sketch:

`#include "Arduino.h"
#include "WiFi.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"
#include "Wire.h"
#include "AC101.h" //https://github.com/Yveaux/AC101
#include "Audio.h" //https://github.com/schreibfaul1/ESP32-audioI2S
#include <MFRC522.h> //https://github.com/madias123/STM32duino-RFID-Mp3-Player/tree/master/3d%20party%20libraries%20(modified)/MFRC522

// SPI GPIOs
#define SD_CS         13
#define SPI_MOSI      15
#define SPI_MISO       2
#define SPI_SCK       14

// RFID
#define MFRC522_RST_PIN                   22
#define MFRC522_CS_PIN                    12
extern SPIClass SPI_MFRC;
MFRC522 mfrc522(MFRC522_CS_PIN, MFRC522_RST_PIN);

// I2S GPIOs, the names refer on AC101
#define I2S_DSIN      25
#define I2S_BCLK      27
#define I2S_LRC       26
#define I2S_MCLK       0
#define I2S_DOUT      35

// I2C GPIOs
#define IIC_CLK       32
#define IIC_DATA      33

// Tasten
// #define BUTTON_2_PIN 13             // shared mit SPI_CS
#define BUTTON_3_PIN  19
#define BUTTON_4_PIN  23
#define BUTTON_5_PIN  18               // Stop
#define BUTTON_6_PIN   5               // Play

// Amp enable
#define GPIO_PA_EN    21

//Switch S1: 1-OFF, 2-ON, 3-ON, 4-OFF, 5-ON

String ssid =     "ssid";
String password = "pwd";

static AC101 ac;
int volume = 40;                                  // 0...100

Audio audio;

//#####################################################################

void setup()
{
  Serial.begin(115200);
  Serial.println("\r\nReset");
  Serial.printf_P(PSTR("Free mem=%d\n"), ESP.getFreeHeap());

  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);
  pinMode(MFRC522_CS_PIN, OUTPUT);
  digitalWrite(MFRC522_CS_PIN, HIGH);
  SPI_MFRC.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPI_MFRC.setFrequency(1000000);
  while (!SD.begin(SD_CS, SPI_MFRC)) {
    Serial.print(".");
    delay(500);
  }

  mfrc522.PCD_Init();
  mfrc522.PCD_DumpVersionToSerial();

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());

  uint8_t tryCount=0;
  while (WiFi.status() != WL_CONNECTED && tryCount <= 4)
  {
    Serial.print(".");
    delay(500);
    tryCount++;
    if (tryCount >= 4 && WiFi.status() == WL_CONNECT_FAILED) {
      WiFi.begin(ssid.c_str(), password.c_str());     // Workaround for ESP32-bug
    }
  }

  Serial.printf_P(PSTR("Connected\r\nRSSI: "));
  Serial.print(WiFi.RSSI());
  Serial.print(" IP: ");
  Serial.println(WiFi.localIP());

    Serial.printf("Connect to AC101 codec... ");
    while (!ac.begin(IIC_DATA, IIC_CLK))
    {
        Serial.printf("Failed!\n");
        delay(1000);
    }
    Serial.printf("OK\n");

    ac.SetVolumeSpeaker(volume);
    ac.SetVolumeHeadphone(volume);
//  ac.DumpRegisters();

    // Enable amplifier
    pinMode(GPIO_PA_EN, OUTPUT);
    digitalWrite(GPIO_PA_EN, HIGH);

  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DSIN);
  audio.setVolume(10); // 0...21

  audio.connecttoFS(SD, "/Led Zeppelin - Stairway to heaven.mp3");
//  audio.connecttoSD("/Banana Boat Song - Harry Belafonte.mp3");
//  audio.connecttohost("http://mp3channels.webradio.antenne.de:80/oldies-but-goldies");
//  audio.connecttohost("http://dg-rbb-http-dus-dtag-cdn.cast.addradio.de/rbb/antennebrandenburg/live/mp3/128/stream.mp3");
//  audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");

}

//-----------------------------------------------------------------------

void loop()
{
    audio.loop();
    if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreaminfo(const char *info){
    Serial.print("streaminfo  ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}`

One important thing: RFID-reading takes place too often. This leads the music to sound ugly. Don't worry, it's just normal for this one.

eazy, 2020-11-14:
Thanks for the code. Awsome, Wifi is working, SDCard mounted and the MFRC is also working.

But, Audio isn't good, maybe half of speed an noisy, I will try another MP3 or stream later on.

Thanks I'll know going for debugging ;-)

eazy, 2020-11-14:
lol .... I've just seen your last sentence ;-) so everything is fine ;-)

biologist79, 2020-11-14:
Hehe.
So finally, did this one run instantly or did you discover a problem with the wiring (or whatever)?

Regarding the "ugly" sound: decoding mp3 seems to be pretty time-critical. So that's finally why I ended up in running rfid, mp3 and neopixel within separated tasks. Because in case the task-scheduling is controlled by freertos which means: not my job! :-)

eazy, 2020-11-15:
it was running instantly. So no wireing problem.
So now I have to find out why the tonuino isn't running with rfid and SD-Card, I'll report if I find out.

eazy, 2020-12-06:
Sorry, there was too much work the last days.

I've done a bit but not that much, mainpart was the resoldering to shorten the wires (just to go for sure).
deaktivated neopixel and all the stuff which I don't need for the moment bus can't get it work.

Your Isoatited snipped is working alle the time but the normal code has the problem with the sd card and the RFID Reader.

Maybe I'll switch to the Hardware setup of your mainproject, just to get anything working with my less time at the moment.

Zip-o-mat, 2020-12-10:
I dont know if we have the same problem here but it seems that the RFID reader only works if the CS pins are set before SPI_MFRC.begin(). The code now looks like this:
```c++
...
pinMode(SD_CS, OUTPUT);
digitalWrite(SD_CS, HIGH);
pinMode(MFRC522_CS_PIN, OUTPUT);
digitalWrite(MFRC522_CS_PIN, HIGH);

SPI_MFRC.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
SPI_MFRC.setFrequency(1000000);
...
```
if @eazy can confirm this, I can create a pull request for this.

biologist79, 2020-12-10:
To be honest: I still need to put work on this audio-kit-stuff. It needs to be branched I guess in order to keep it in sync with master.

eazy, 2020-12-11:
@Zip-o-mat I'll try out this evening

eazy, 2020-12-11:
@Zip-o-mat with your changes SD Card is mounted and working but RFID is only initialized but not reading any rfid card.
Just no reaction if I put an rfid chip on the reader.

biologist79, 2020-12-11:
Have you been using the Lib provided? https://github.com/biologist79/Tonuino-ESP32-I2S/tree/master/Hardware-Plaforms/ESP32-A1S-Audiokit/lib/MFRC522
I know somehow this should work with the standard-lib but I didn't have time yet to investigate.

Zip-o-mat, 2020-12-11:
I just double checked:
This is the only thing I changed in the source code and I am using the modified MFRC522 Library (I commented out the original Library in my platformio.ini ( ; https://github.com/biologist79/rfid.git ))

eazy, 2020-12-11:
yes, I'm also using the modified one. I also commented the original one from platformio.ini and to go for sure I've done the include with the complete path
#include <..\lib\MFRC522\src\MFRC522.h> 

and I tested also the "isolated sketch" to go for sure wiring is ok, this is working

eazy, 2020-12-11:
urgs ... I erased the esp and now it is working with the codechange of @Zip-o-mat 

thanks

biologist79, 2020-12-11:
So please send me a pull-request.
However, still need to integrate that stuff into my main-development. The code in audiokit's folder is somewhat outdated.

closed 2020-12-16 by biologist79