biologist79/ESPuino #10: File Browser for RFID-assignments

mariolukas, 2020-11-28:
Just want to let you know that i am working on a File Browser for making RFID-assignments more convinient.

[image]

I will do a pull request when it's done. I am still struggling with the json formating.

biologist79, 2020-11-28:
Wow, that is really really cool! Indeed my initial plan was EXACTLY the same (*) but I had so much work with all that other stuff.  And finally, to be honest, Js is not my world 👍 So I really appreciate your plan/work. The way to use a pre-formatted json makes absolutely sense. Just tell me if you need support (apart of Js, hehe).

(*) Indeed I had some out-commented code that was used for parsing (maybe you recognized). However, I removed it a few days back as I thought this feature will never be released :-D

biologist79, 2020-11-28:
BTW: Can't wait to see this running on my Tonuinos! :-D

mariolukas, 2020-11-30:
It is almost finished. I will do some final tests and finally do a pull request.   

[image]

biologist79, 2020-11-30:
Cool!
A second thing: When doing RFID-linking, a green box opens to indicate a successful operation. I'd like to automatically dismiss this box after a few seconds (3s or so). Tried around a bit but in the end it didn't work. I was able to automatically dismiss it but after that it was dismissed it never appeared again for RFID-linkings done subsequently. Guess it's not a big thing; maybe you know how to do that.

mariolukas, 2020-11-30:
What do  you think about using [toastr](http://plnkr.co/edit/6W9URNyyp2ItO4aUWzBB?p=preview) messages for indicating successful operations, warning,  errors etc.
I am  already testing the file browser indications using toasts.

[image]

biologist79, 2020-11-30:
In German I'd say: "Ich bin da emotionslos" :-) So: I'm fine with it.
Personally I just make sure to use my own server-infrastructure (for link-hrefs) in order to avoid monitoring/tracking by Google & Co.

mariolukas, 2020-11-30:
> In German I'd say: "Ich bin da emotionslos" :-) So: I'm fine with it.

Passt. 

> Personally I just make sure to use my own server-infrastructure (for link-hrefs) in order to avoid monitoring/tracking by Google & Co.
So you just need to copy the needed libs to your server and replace the links by your own.

biologist79, 2020-11-30:
Yes, I know. I also did this in the past for bootstrap and jQuery as well.
However, had a look at https://codeseven.github.io/toastr/demo.html and looks really good I have to admit. So I'm really looking forward to your pull request :-)

mariolukas, 2020-12-02:
feature/filebrowser #12 
I tried two times to create a pull request which  contains only the file browser changes. I merged with the upstream master before i created the pull request, maybe that is the reason why so many changes are applied. Not the best way to understand what i changed. The additions to the main.cpp are mainly new functions for the file index handling and json writing, some websocket event handling stuff and a static route for delivering the files.json. 
I am using anther esp32 board, that is the reason why i changed platformio.ini (but added a platformio.ini.default file). I would appreciate moving all the config/defines to an extra file like config.h or something similar. I think that makes it easier for us all to handle custom pinouts without messing up the code with every pull request. I still have some additional ideas but will open new issues for those :)

biologist79, 2020-12-02:
Definitely outsourcing stuff makes totally sense; 100% ACK! Will be one of the next things I'll change.
Now I don't know how to merge all that stuff to my code to be honest because the left files in the diff are not my current master. Wouldn't be a problem to do it manually as I know where to look, but then the info is gone that you're the contributor. Hmm.

mariolukas, 2020-12-02:
I will give it a try with a fresh fork. I think that would be the easiest way for both of us.

biologist79, 2020-12-02:
Couldn't wait and c&p mgmt-html-File and flashed it onto my Lolin32. As I don't have an json-file on the SD-card, the GUI keeps "loading". However, every 10-15s (or so) a msg appears "Aktion erfolgreich ausgeführt" even if I don't do anything.
Addtionaly you stated "margin-botton" in the html-head instead of "margin-bottom". 

Haha, that's a new experience for me to assign bugs to anybody else in my own project :-D However, big thanks for your contribution!

mariolukas, 2020-12-02:
> Couldn't wait and c&p mgmt-html-File and flashed it onto my Lolin32. As I don't have an json-file on the SD-card, the GUI keeps "loading". However, every 10-15s (or so) a msg appears "Aktion erfolgreich ausgeführt" even if I don't do anything.

Yes the spinner is running, cause it tries to load a file. I think this is caused by the tree library. I will try to catch the 404 error message. See comments in pull request (todo). 

First time you need to hit the refresh icon below the box for initial index file creation. This might take some time, depending on how many files are on your card.  

"Aktion erfolgreich ausgeführt" 
Something what i also discovered. But this is related to the ping/pong event.  I was too lazy to have a closer look to the websocket routine where the "ok" event is triggered. But it seems to be triggered every time when a pong is received. You can discover that on the js console. I think this behaviour was not visible before, cause the old notification was more a static approach. 

> Addtionaly you stated "margin-botton" in the html-head instead of "margin-bottom".

Thanks.

> 
> Haha, that's a new experience for me to assign bugs to anybody else in my own project :-D However, big thanks for your contribution!

So you finally managed merging this?

biologist79, 2020-12-02:
Clicking the refresh-button would not have been leading to a solution as I (really) only did c&p with the html-file. So no server-support.
So: no, didn't manage the merge. I'm not super-familiar with git, so would have to point out. However, doing it manually would not be a problem, but then nobody knows it was your contribution. Ehre, wem Ehre gebührt :-)

mariolukas, 2020-12-02:
Don't worry. I will try to fix that merging mess and will do another pull request.

mariolukas, 2020-12-03:
I think i managed the merging problem. But i will fix some minor bugs before i will do the pull request. The following todos are open: 

- notification when no file list is available
- more stateful nofitications during file indexing, this can take a long time depending on the number of files

biologist79, 2020-12-03:
Looks good, thank you!
Will try this tonight I guess.

biologist79, 2020-12-03:
Hmm, created a new branch, did a pull-request by myself and synct it locally to a branch. Ok so far but it's about time to gain knowledge on git :->
However, did some small changes in platformio.ini, compiled and flashed the stuff. Tonuino + GUI is working, but GUI doesn't interact with Tonuino (even ping/pong) etc. Have to point out what's wrong. Firefox' debugger told me that a .js-File was not downloadable from my server (which is crap; it works, I tested it manually).

Will have a look later.

mariolukas, 2020-12-03:
Maybe there are still hard coded ip addresses in the html, which i added for debugging. I managed to make it conditional in the current version which is still not commited.

Give me some time. This branch was even not final. I did a lot of additional work on it which was still not commited and pushed to GitHub. I will push the latest changes tonight. I am not even sure if there is a need for the ping/pong stuff by using websockets. The websocket connection is hold on until it will be actively closed by the browser or by the tonuino (what also happens when it runs into a timeout).

Anyway I started working on this issue because i am having a problem when the file indexing function is called. Indexing works like a charm, but it blocks the websocket process from sending ping/pongs, what leads to a lot of messages in the frontend, even if the websocket connection is still alive, cause the (ping/pong) timer in JavaScript is doing its job well :)

 I already tried to add a couple of watchdog timer resets during the recursion while file indexing. But without success. This keeps the watchdog running but the websocket task is still blocked. In addition to all that i also tried making a task on the second cpu of all the filebrowser stuff, without any luck so far.

biologist79, 2020-12-03:
Ok, take your time 👍 
It's only in my devel-branch so there's no need to rush.

mariolukas, 2020-12-03:
Maybe i am not able to fix this socket issue by myself. It is really hard to debug and i already looked over most of the issues on asyncWebServer library. I hope i will find a solution.

biologist79, 2020-12-03:
Commit what's working and I'll see what I can do.

biologist79, 2020-12-03:
Regarding task-stuff: did you have a look at my tasks? I used vTaskDelay() along with esp_task_wdt_reset(). For example if the music stops, it has to be called as the loop runs too fast without yield(). But indeed, that's not so easy - I also spent a lot of time to get it running as it's (in my opinion) not very well documented. I've been missing something like a "guideline for task-programmers", hehe.

biologist79, 2020-12-04:
Wasn't aware you already commited. However, had I look at your repository and merged it into my devel-branch + tested it. Is really looking and working great! So finally, have you been successful in using vTaskDelay() or are there still problems ongoing? However, I didn't notice reboots while creating the json-file. The only thing I noticed was like it took three times or so to create the json-file when starting for the first time. However, not a big thing. Guess will have a look at it this evening.

Like it! Especially learning RFIDs via cellphone is nice!

mariolukas, 2020-12-04:
I added my latest changes a couple of minutes ago: 
https://github.com/mariolukas/Tonuino-ESP32-I2S/tree/feature/sdbrowser

Indeed i managed to fix the issue with the blocking stuff. However as you discovered we have one thing left open. We need to find a solution to prevent multiple task creation when the indexing task is already running. The task is not like the others, it does not run in  a loop. It runs and terminates itself after execution. And that is exactly the point where we need to add a mutex or semaphore, to prevent multiple task execution.

https://github.com/mariolukas/Tonuino-ESP32-I2S/blob/f940a2ee6736eb70526cb1798cded4c7f2d7fc91/src/main.cpp#L3309

mariolukas, 2020-12-06:
I think this is finally done.

closed 2020-12-06 by mariolukas