biologist79/ESPuino #15: i18n Support for the user interface.

mariolukas, 2020-12-04:
I think it would be nice if there is a more elegant way for handling i18n. In my opinion the serial debug (backend) translations are not so important. More important is the user interface stuff. A more elegant way for handling translations would be using a JavaScript i18n Library. I know more JavaScript stuff :) But let the heavy stuff happen in the frontend where the Browser can handle the mighty heavy String handling stuff. 

My idea is: 
1. writing the HTML in a default language (maybe german or english)
2. using a i18n library which is able to load dynamically json language files from the SD-Card.
3. when no file is found (initial state) the default language is uses
4. choosing the language can be set depending on the used browser language or by selecting it manually in the user interface.

However this will bring the benefit that even people without programming skills can commit something to the project, furthermore there is less static code in header files and only one html file needed to edit.

biologist79, 2020-12-05:
100% ACK. Nothing more to say :-)

mariolukas, 2020-12-06:
Just spent some time in evaluating the[ jquery i.18n](https://github.com/wikimedia/jquery.i18n) library. It seems exactly to be what i was looking for. Additionally it is well maintained, cause it is part of the wikimedia project. I think thats the best i18n framework for this project.

mariolukas, 2020-12-06:
This is in a really early state but it can already be tested. I started a new feature branch for i18n stuff and refactored a couple of things related to the html behaviour (renamed files, removed EN language files because those will be replaced by the new i18n feature, added some css for a nicer look of the initial access point page, refactored the python script for header file and html processing).

The idea is to create a folder on the sd-card, called i18n. The json files in html/i18n/ should be placed there. The user interface checks for the browser language and loads the correct language file. When no file was found or a file was not loadable, the english will be used as fallback.

biologist79, 2023-02-07:
Finally solved with https://github.com/biologist79/ESPuino/pull/194
Yeah!

closed 2023-02-07 by biologist79