biologist79/ESPuino #19: Wifi problem: Esp doesn't get proper IP assigned

jokober, 2020-12-06:
Hello, 
first I want to thank you for your great work!

Unfortunately I am struggling to get everything to work. So far I only have the sd card module and the rfid reader connected. I had some issues to get the sd card reader to work with my Lolin32 Lite (no 5V pin) but managed to solve the issue by using a external 5V source to power the sd card module. This is not a nice solution but it works for the moment.

After connecting to the access point, changing the wifi settings and restarting the esp is not properly connected to the network. I can see the device in the fritz!box settings but it has no IP assigned. DHCP is aktivated.

The output also looks really weird:

```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
Hostname aus NVS geladen: jettebox
Firmware Version: 0xEF = (unknown)
RFID-Tags koennen jetzt gescannt werden...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:418] _eventCallback(): STA IP: 255.255.255.255, MASK: 255.255.255.255, GW: 255.255.255.255
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:418] _eventCallback(): STA IP: 255.255.255.255, MASK: 255.255.255.255, GW: 255.255.255.255
.Aktuelle IP: 255.255.255.255
SD Card Type: SDHC
Free heap: 153464
Versuche Verbindung zu MQTT-Broker aufzubauen 192.168.2.43
Verbinde zu MQTT-Server mit User und Passwort
Neue Lautstärke empfangen via Queue: 3

```

Have you experienced similar problems or do you have any idea what might cause the problem?

biologist79, 2020-12-07:
Hi,

hmm, I've never been working with a Lolin32 lite. However, I've been told that this boards makes use of a weak voltage regulator. The user said he wasn't able to get the SD-card running until he used an external voltage-regulator. So (to be honest) just a rough guess: maybe a problem due to the regulator?

Have you already been using this board with a simple Wificlient-sketch or so? Just to make sure it's working in this case.

mariolukas, 2020-12-08:
I am not using a Lolin32 Board but sometimes i am running in a similar issue. I have the presumption that MQTT might bring some problems when no broker is configured in your network. Maybe you can try to deactivate MQTT for a first try. Just change your config file from: 
`#define MQTT_ENABLE   `  
to 
`//#define MQTT_ENABLE      `

biologist79, 2020-12-09:
@jokober Have you been testing my latest software-revision? Just asking because I revised MQTT-handling recently. Maybe your problem is resolved now.

jokober, 2020-12-09:
Sounds promising. I will be able to test it tomorrow evening. Thanks for the help

jokober, 2020-12-11:
Disabling MQTT as well as the new software-revision did not fix the problem. Unfortunately I don't have any new information which might help to solve the issue. I will let you know if I sind out more

jokober, 2020-12-13:
I managed to solve the problem. It was kind of obvious but still took me some time. First I realised that I could connect to the network with a static IP. 

Next I made sure to connect to the network by removing following line:

https://github.com/biologist79/Tonuino-ESP32-I2S/blob/38556b5d69a39d133b05e957f4475278b675e82d/src/main.cpp#L2967

And voila it works!

closed 2020-12-13 by jokober

biologist79, 2020-12-13:
Thanks you! Fixed it.