biologist79/ESPuino #25: Rotary Encoder - no INPUT_PULLDOWN possible

erniberni, 2020-12-10:
I am not sure if this really matters.
The GPIO34 and GPIO35 where the rotary encoder is connected has no sw enable pullup or pulldown.
The used library ESP32Encoder default enables a pulldown on these GPIOs but this is not working.

biologist79, 2020-12-11:
Didn't have any problems so far. Encoder's button is pulled internally (in the encoder) to high if not pressed.

tueddy, 2020-12-11:
I can confirm the encoder works without any problems.

closed 2020-12-11 by biologist79

tueddy, 2020-12-18:
This could be the reason why it works flawless:

[image]

If you put just the rotary encoder on a PCB you should use external pullups. For the breakout module it works because resistors are already on the module.

erniberni, 2020-12-20:
Yes, I am sure, this is the reason. I am using an encoder like this https://www.amazon.de/Rotary-Encoder-Schalter-Drehschalter-Keyswitch/dp/B0085I4D5C/
That type has no pullups. I used an external pulldown, because the library ESP32Encoder is expecting pulldowns as default. I think this doesn't really matter, if the sw is detecting signal edges.