biologist79/ESPuino #28: SPI configuration on ttgo t8 v1.7 (SD card reader onboard)

jokober, 2020-12-14:
I am trying to use the ttgo t8 v1.7 in my project. It has an onboard card reader. Unfortunately I can't get the RFID reader (RC522) to work. I saw that you @biologist79  have played around with different SPI configurations on different boards and was wondering if you have an idea what I do wrong or what I could try to get the RFID reader work. Previous to this project I had no experience with SPI.

Here is my configuration:

```
//#define SINGLE_SPI_ENABLE         // If only one SPI-instance should be used instead of two (not yet working!)
// uSD-card-reader (via SPI)
#define SPISD_CS                        13 // <- changed from 15          // GPIO for chip select (SD)
#ifndef SINGLE_SPI_ENABLE
    #define SPISD_MOSI                  15 // <- changed from 13            // GPIO for master out slave in (SD) => not necessary for single-SPI
    #define SPISD_MISO                 2 // <- changed from 16             // GPIO for master in slave ou (SD) => not necessary for single-SPI
    #define SPISD_SCK                   14          // GPIO for clock-signal (SD) => not necessary for single-SPI
#endif

// RFID (via SPI)
#define RST_PIN                         99          // Not necessary but has to be set anyway; so let's use a dummy-number
#define RFID_CS                         21          // GPIO for chip select (RFID)
#define RFID_MOSI                       23          // GPIO for master out slave in (RFID)
#define RFID_MISO                       19          // GPIO for master in slave out (RFID)
#define RFID_SCK                        18          // GPIO for clock-signal (RFID)
```
The ttgo t8 board:
[image]

tueddy, 2020-12-14:
Hi,

have you checked SD card reader can mount the card?
There might be a conflict with
`#define PREVIOUS_BUTTON                 2`
IO2 is used for MISO in your configuration. 
So the software crashed before the RFID reader could be started.
What is the output in serial console?

jokober, 2020-12-14:
Good point. I have changed the PREVIOUS_BUTTON pin.

However it doesn't change anything. The SD card can be mounted. There is no output when scanning a RFID tag:
```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
Firmware Version: 0x92 = v2.0
RFID-Tags koennen jetzt gescannt werden...
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:418] _eventCallback(): STA IP: 192.168.178.68, MASK: 255.255.255.0, GW: 192.168.178.1
.Aktuelle IP: 192.168.178.68
Free heap: 163664
Neue Lautstärke empfangen via Queue: 3
Batteriespannung niedrig: (0.30 V)
Aktuelle Batteriespannung: 0.30 V
```
After some time following output appeared:
`31169 BAIL 12`

jokober, 2020-12-14:
At least I thought the SD card works. In the web-gui I get following notification:
"SD-Karten-Index muss erstellt werden"

biologist79, 2020-12-14:
Your SD-card needs to be indexed in order to provide the filebrowser. This is because reading from SD all the time is a bit slow. But means: every time you changed anything on the SD-card, the indexfile needs to be refreshed (only takes a few seconds).

Did you connect the RFID-reader to 3.3V?
Try to get RFID only running: https://github.com/miguelbalboa/rfid/blob/master/examples/DumpInfo/DumpInfo.ino
RST is not necessary. SS is CS. So should be necessary to change SS to 21

tueddy, 2020-12-14:
Strange, `Firmware Version: 0x92 = v2.0` means the MFRC522 is wired up fine and running.

biologist79, 2020-12-14:
Indeed I had such problems with RFID not complaining but anyhow not running. But only with single-SPI-mode. With two SPI-instances it is fine.

jokober, 2020-12-14:
I'm sorry the problem was caused by a faulty breadboard. I can confirm that the above configuration works on the ttgo t8. Now everything works well from RFID scan to the audio playing. Thank for your great work!

biologist79, 2020-12-15:
Sounds good :-) I often had problems because of that breadboard-stuff. That's why I started to create PCBs. But it totally makes sense to test this stuff first :-)

closed 2020-12-15 by biologist79