biologist79/ESPuino #31: gerber files for D32 missing outline

michi1g, 2020-12-17:
I tried to order your PCBs form jlcpcb as you suggested but after review they sent me an email:
```
Audit Failed
Failure Reason:
1.Hi Sir/ Madam, there is no board outline in your file, please kindly check it!
```
Can your recheck the board?

biologist79, 2020-12-17:
Hi,

sorry. I revised the PCB. Please try again.

Greets.

michi1g, 2020-12-17:
Thx for the fast help. I reuploaded and will close the issue if it passes

michi1g, 2020-12-17:
Unfortunately still getting the same error from jlcpcb

biologist79, 2020-12-17:
Hmm. Omitted dimensioning this time. Maybe now?
Sorry...

biologist79, 2020-12-20:
So? Working now?

closed 2020-12-21 by biologist79

biologist79, 2020-12-29:
As it turned out it was still missing. Outline was on wrong layer; sorry!
However, should be fixed now (finally!).