biologist79/ESPuino #35: No RFID tag via MQTT

trewbene, 2020-12-21:
Thank you very much for this great project!

The days I came up with another idea, what I would like to implement with the box:
To help our daughter learn colors, I painted her RFDI cards with colors (red, green, blue, etc.).
When she puts the card on, a track is played to her that explains what the color is called.

Now I have the idea that I could link the whole thing via MQTT with the light chain on the Christmas tree.
Speak:
I put on a card, get the ID via MQTT, which I use in HomeAssistant or NodeRed to change the color of the lights on the Christmas tree.

Yesterday I noticed a little something:
According to the documentation on the home page "topicTrackState" should send the following: "Sends number of last RFID-tag applied".
That would be great and would work perfectly, since the ID is always unique for each card.
Instead, however, the current track is published via MQTT, which is not always unique due to random modes, etc.
Would it be possible to publish two different topics for the title and the rfid tag?

biologist79, 2020-12-21:
Hehe, sounds interesting.
Indeed, you're right: I've made a mistake in my documentation. The topic contains something like "(#/#) <trackname>". Already changed it. I'll implement sending RFID-ID via MQTT; guess this evening. Stay tuned :-)

biologist79, 2020-12-21:
Should work now.

biologist79, 2020-12-21:
Please look at https://github.com/biologist79/Tonuino-ESP32-I2S/commit/ae8aac8e0fd38b6afc87deeb6812d40e4966644a in order to know what has changed.

closed 2020-12-21 by biologist79

trewbene, 2020-12-22:
Wow, thanks for the quick implementation!
I'll try it tonight and give feedback.

trewbene, 2020-12-22:
Perfect, now works as planned. Thanks a lot!