biologist79/ESPuino #36: Uploaded files via FTP are always empty

weitzj, 2020-12-21:
I have tries to upload files via the Linux `ftp` command or via filezilla with a single connection and `passive` disabled.
But the files are just `touched` on the SD-card, but no content is written to the file.

Do you have any tips, what I am doing wrong?

weitzj, 2020-12-21:
Ah. I think i am getting closer:

```bash

# open ftp client
ftp

# connect to your Tonuino
# username, password are esp32
open 192.168.178.73

# enable passive mode
passive

# send a file 
send myfile.mp3 

```

weitzj, 2020-12-21:
The working setup. Filezilla with 1 connection in passive mode:

[image]

[image]

weitzj, 2020-12-21:
Thank you for this great project :heart:

closed 2020-12-21 by weitzj

biologist79, 2020-12-21:
What I can add from my perspektive: I had issues where FTP-transfer seemed to take place (round about twice as fast as usual), but in the end file's size was 0. I guess this was due to bad wire-connection(s) on the breadboard.