biologist79/ESPuino #37: LED Rainbow hue does not seem to work?

weitzj, 2020-12-21:
I have hooked up an LED ring and it mostly works, i.e. 

- white if bootd with wifi (animated)
- blue if started without wifi (animated)
- orange during boot (animated()

But I can't get the rainbow hue working whilst playing a single MP3 file.
When playing a single file the LEDs turn off.

When playing a webstream I get 2 opposite red LEDs (not animated)

My question is:

Is this intended behaviour? (to safe battery)

If I were to implement it, do I need to create another LED state like `ledStatusOk`, `ledStatusError`?

weitzj, 2020-12-21:
I see you have upgraded the master. I will try before openning this issue

closed 2020-12-21 by weitzj

biologist79, 2020-12-21:
Currently, as it seems, there are some problems ongoing with the lib I used (Fastled). Recently had serveral restart-issues which is the reason I'm now back at version 3.3.3 for Fastled (see platformio.ini) as reference. So I don't know if that might be reason (but probably not).

However, when it comes to Neopixel, I didn't implement anything different for saving energy. In fact ESP32 doesn't know (and so me) if battery currently is the only source. Well, I could if I'd do an analogRead() via voltage divider of a 5V-pin, but as it's never been my intention.

I never had reports so far, that rainbow-color doesn't work for track-progress. The only thing I noticed is: if brightness is down to 2 or so, colouring changes a bit. Does the rainbow appear/disappear when changing the volume via rotary encoder?

biologist79, 2020-12-21:
Today I only updated README-description.

weitzj, 2020-12-21:
Thank you.

The expected behaviour for webradio is:

2 red leds opposite of each other rotating counterclockwise? And once in a while a green blink?

biologist79, 2020-12-21:
2 leds: yes. But when I remember right, rotation is clockwise (not 100% sure).
However, in settings.h there's a switch available to invert LED-addressing as there are Neopixel around soldered clockwise wheras others are counter-clockwise.

biologist79, 2020-12-21:
Green blink: No. This is a problem with version 3.3.3. Don't have these green blinks with the master-version of Fastled. But this version led to restart-issues. So green blinks is "das kleinere Übel" :-)

weitzj, 2020-12-21:
Sorry for my dumb questions. RTFM 

The LED Ring does a progressbar (with green blink once in a while). Now everything works!

Fantastic project