biologist79/ESPuino #38: Wifi AP reset functionality

weitzj, 2020-12-21:
Is it possible to implement a mode, where I can create an RDID mapping + content on the SD card but reset my local Wifi configuration?

i.e. I want to prepare this Box as a present but I don't know the SSID/PW of the person who gets the present.

Is this maybe already somehow implemented?
How would you suggest to tackle this problem?

- push and hold 3 buttons for 10 seconds to reset wifi?
- have a special RFID card to reset wifi?

biologist79, 2020-12-21:
Sorry, but I guess it's indeed time for RTFM :-)
Tonuino tries to connect to WiFi. If that's not possible, it opens an access-point that you can connect to (IP: 192.168.4.1). After entering Wifi-credentials there + restart, Tonuino joins this WiFi configured. So you don't need to know your friend's WiFi-credentials.
WiFi can be switch on/off with two buttons (prev + next).

weitzj, 2020-12-21:
Do you have a paypal address, so I can send you a small christmas present?

closed 2020-12-21 by weitzj

biologist79, 2020-12-21:
Wrote you an email :-)