biologist79/ESPuino #46: Support of Bluetooth headphones

datenzar, 2021-01-05:
Instead of using the Headphone Jack, the ESP32 could also stream the sound to Bluetooth headphones. Unfortunately according to [this ticket](https://github.com/espressif/arduino-esp32/issues/3558), it's not integrated into the ESP32 Arduino library but maybe someone has an idea. If I find the time, maybe I will work on it myself.

Feedback welcome!

biologist79, 2021-01-05:
It should work with this library: https://github.com/pschatzmann/ESP32-A2DP.git
So far I've just been testing a2dp-sink and it turned out it's critical because auf lack of memory. What could (probably) improve this situation is to use PSRAM (available on boards that make use of ESP32-WROVER). But to be honest I'd prefer to have a solution that runs on ESP32-WROOM as well. When I was out for a run last week I had an idea: adding a new modification-card that toggles between "normal" output and a2dp-sink (or in your case stresm-server, respectively). However, would be necessary to delete+free the instance dynamically, that's currently not used.

So far it's just an idea. Didn't put any efforts on it, yet.

datenzar, 2021-01-05:
Thanks for the link, I will check it out!

You're right regarding memory. Either work hard on using it most efficiently or use a board with more PSRAM. I will test the library when I find time.

biologist79, 2021-01-05:
At least as a2dp-sink, it runs really fine. Just did a test yesterday and all code, that was necessary, is:

`#include <Arduino.h>
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

#define I2S_DOUT 25
#define I2S_BCLK 26
#define I2S_LRC 27

void setup() {
  i2s_pin_config_t pin_config = {
    .bck_io_num = I2S_BCLK,
    .ws_io_num = I2S_LRC,
    .data_out_num = I2S_DOUT,
    .data_in_num = I2S_PIN_NO_CHANGE
  };

  a2dp_sink.set_pin_config(pin_config);
  a2dp_sink.start("a2dp-sink");
}

void loop() {
}`

biologist79, 2021-02-09:
BTsink is fully integrated now. Guess "your feature" needs support by the creator of the bt-lib. Here's an interesting discussion where he's been involved: https://github.com/pschatzmann/ESP32-A2DP/issues/11
I don't use ESP8266Audio but in the end should be the same.

closed 2021-04-25 by biologist79