biologist79/ESPuino #51: SD could not mounted

thebino, 2021-01-10:
I'm trying to use [this SD-Modul](https://de.aliexpress.com/item/32523546123.html) but the SD-Card could not be mounted.
> [E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)

Do I need to use the **SD-MMC Mode**?

As ESP I use the [ESP32-WROOM-32 from AZ-Delivery](https://www.amazon.de/gp/product/B074RGW2VQ/)

tueddy, 2021-01-10:
I think this module is for 5V boards. There is a power regulatur, so Vcc must be 5V.
I've tested this board before, works fine with SPI mode. You have to power with 5V or better remove the power regulator abd bridge them. Then you can use the ESP-32 3.3V for Vcc:

[image]

I don't remember if this board works in SD-MMC mode (I don't think so).

thebino, 2021-01-10:
Ah, I've read something about this power regulation change but didn't check the spec for this module which resistor I have to remote.

tueddy, 2021-01-10:
If you want to test SD-MMC mode on a breadboard, you can solder this simple and cheap Adapter:

[image]

I can recommend this [article ](https://www.instructables.com/Select-SD-Interface-for-ESP32/)comparing SPI and SD-MMC modes.

thebino, 2021-01-10:
[image]

Nothing changed, can I measure anything to check the voltage?

biologist79, 2021-01-10:
Moin,

ich wechsele mal kurz auf Deutsch. Wir müssen uns als drei Deutsche hier auch nicht auf Englisch unterhalten :-)
Habe mir die Platine von Mario eben mal in KiCad angeschaut. Denke hier muss ich die Doku noch etwas erweitern. Hier ist ja ein Mosfet und ein NPN-Transistor eingebaut, der die Peripherie abschaltet, wenn der ESP im Deepsleep ist. Allerdings ist es so, dass beim Board von Mario 5V geschaltet werden und nicht 3.3V, wie ich das inzwischen (aus Akku-technischen Gründen) verstärkt propagiere. Ausnahme ist der RFID-Reader, der hängt direkt am GPIO17, der sonst die Transistorstufe ansteuert. Das bedeutet, dass der SD-Reader mit 5V versorgt wird. Da SD nur mit 3.3V arbeitet, braucht es hier einen Festspannungsregler. Den hat das von dir verwendete Board eigentlich auch (habe es selbst hier auch liegen), aber er ist ausgelötet. Warum? 
Teste am besten mal im Rechner, ob deine SD-Karte überhaupt noch funktioniert.

tueddy, 2021-01-10:
Kenne jetzt diese Platine/Layout von Mario nicht, was hier wie wo beschaltet wird. Der Vorschlag den Spannungsregler auszulöten bezog sich auf die direkte Ansteuerung mit ESP-32 über 3.3V. Das Breakout Board ist für 5V ausgelegt. Wenn das mit 5V belegt wird bleibt der Spannungsregler natürlich drin..

biologist79, 2021-01-10:
Ach das habe ich übersehen
@thebino Also der Festspannungsregler muss auf jeden Fall rein. Wo du aber auf jeden Fall ansetzen kannst, das sind die Lötpunkte. Achte drauf den Pin, den du festlöten willst, vielleicht erst so 3s mit einem Lötkolben zu erhitzen. Dann gibst du *etwas* Lötzinn dazu und nimmst das Lötzinn wieder weg. Dann wartest du noch kurz und wirst sehen, dass das Lötzinn anfängt zu fließen. Dann nimmst du den Lötkolben erst weg und lässt es abkühlen.

Schau mal hier so bei 6min, da kannst du das gut sehen: https://www.youtube.com/watch?v=Cww1ZGKClXw

biologist79, 2021-01-10:
So, ich habe die Doku mal erweitert/verbessert: https://github.com/biologist79/Tonuino-ESP32-I2S/tree/master/PCBs/ESP32_Breakout_Carrier

thebino, 2021-01-11:
Danke ich hätte vielleicht dazu schreiben sollen, dass ich hier ein Custom PCB verwende. Mit ist später noch aufgefallen, dass die Platine überall auf 5V ausgelegt ist.
Die Lötpunkt sehen deshalb so scheiße aus, weil meine Lötsptitze nach ca. 20 Jahren eindeutig hinüber ist, die neue kommt allerdings erst diese Woche. Dann wird nochmal alles ordentlich gemacht.
Die fehlenden Komponenten folgen dann ebenfalls, die kommen nur nicht direkt auf das Board sondern irgendwo in das Gehäuse wo sie Sinn machen.
Ich wollte nur erstmal den SD-Reader testen und dachte das sollte auch ohne RFID gehen bevor ich noch mehr verbaue.

biologist79, 2021-01-11:
Das ist auch tatsächlich der Fall: SD ohne RFID geht. Nur andersrum nicht :-) Weil es hängt inzwischen so viel von SD ab, dass es ohne nicht mehr geht. Es wird demnächst vermutlich demnächst noch eine Anpassung geben, dass man html-Files vor der Einrichtung auf die SD-Karte kopieren muss.

erniberni, 2021-01-13:
Wenn du die SD Karte zunächst alleine testen willst (ich habe das gemacht, um andere Fehler auszuschliessen), dann empfehle ich dieses Testprogramm, das auf die Verdrahtung im Projekt abgestimmt ist.

[SD_Test.zip](https://github.com/biologist79/Tonuino-ESP32-I2S/files/5808717/SD_Test.zip)

thebino, 2021-01-13:
Habe den Fehler gefunden, hatte falsche Widerstände verbaut wodurch die Spannungsversorgung nicht gepasst hat.

closed 2021-01-13 by thebino