biologist79/ESPuino #56: FTP Problems

erniberni, 2021-01-13:
Hallo,
es gelingt mir nicht, mehrere Files oder ganze Verzeichnisse per ftp zu übertragen. Einzelne Files nacheinander klappt.
Ich nutze FileZilla auf Ubuntu, nur damit ist es mir gelungen überhaupt eine ftp-Verbindung vernünftig aufzubauen, da man TLS gezielt ausschalten/verhindern muss.
LG Reinhard

biologist79, 2021-01-13:
Moin,

also tatsächlich habe ich auch das Problem, dass es zu Verbindungsabbrüchen kommt, wenn ich mehrere Dateien in einem Ordner habe (alle 3-4 Files oder so). Filezilla baut die Verbindung aber immer wieder neu auf, insofern habe ich dem noch keine größere Bedeutung zukommen lassen :-)

erniberni, 2021-01-13:
Ja, aber nach mehreren Versuchen wird dann abgebrochen. Ich hatte dann auch schon Dateien, die nicht vollständig übertragen wurden.

erniberni, 2021-01-13:
Gerade mit FileZilla auf Win10 getestet -> gleiches Verhalten

biologist79, 2021-01-13:
Muss man vielleicht mal mit anderen FTP-Libs testen. Sowas zB, wobei die laut Beschreibung nur SD_MMC kann:
https://github.com/fa1ke5/ESP32_FTPServer_SD_MMC
Aber grundsätzlich scheint das alles mehr oder weniger den gleichen Ursprung zu haben.

erniberni, 2021-01-13:
FileZilla kann gleichzeitig mehrere Files übertragen und das mag der ESP nicht.
Im FileZilla unter “Bearbeiten” / “Einstellungen“ / “Übertragungen”. Hier einfach unter “Maximale Anzahl gleichzeitiger Übertragungen” den Wert auf 1 setzen. 
Nach erstem Test scheint das nun mit der Übertragung eines kompletten Verzeichnissen zu funktionieren.

biologist79, 2021-01-13:
Ach ich dachte das hattest du schon so stehen. In meiner Doku steht:
"Make sure to set the max. number of parallel connections to ONE in your FTP-client and the charset to ISO8859-1." Sollte ich vielleicht noch aufnehmen, dass auch keine Verschlüsselung geht.

Der Punkt ist halt, dass die Gesamtanzahl der Verbindungen vom ESP32 limitiert ist. Wenn man das mit FTP vollmacht, dann ist das nicht so zielführend. Insofern finde ich das ok, wenn die Lib das auf 1 limitiert (auch wenn ich das nicht gemacht habe).

erniberni, 2021-01-13:
PUUH, man sollte sich auch wirklich alles - ALLES - am besten mehrmals durchlesen. 
Vielleicht sollte es einen Katalog der wahrscheinlichsten Fehler für so Deppen wie mich geben. ;-)
Ja, das mit "TLS ausschalten" solltest du aufnehmen.
Danke

biologist79, 2021-01-27:
Ich habe gestern mal Durchsatzraten gemessen. Und zwar einerseits mit SPI und auf der anderen Seite mit SD_MMC. Zusätzlich noch FTP und HTTP-Upload (den es ja seit kurzem auch gibt) im Vergleich. Dazu habe ich eine Datei mit 70,7 MiB 

SD_MMC:
FTP (up): 3:57min => 298 kiB/s
FTP (down): 3:45 => 314 kiB/s
HTTP: 3:10min => 372 kiB/s

SPI:
FTP (up): 6:38 => 178 kiB/s
FTP (down): 5:38 => 208 kiB/s
HTTP: 6:24min => 184 kiB/s

So gerne ich FTP auch mag (deswegen habe ich es ja eingebaut): Da muss ich fast eher zu zu HTTP-Upload raten :-)
Da wird es in Kürze übrigens auch noch ein Update geben, welches den Filebrowser und die RFID-Zuweisungen verschmelzen lässt (der Pull-Request ist ja hier schon anhängig).

closed 2021-07-02 by biologist79