biologist79/ESPuino #58: SD Karte wird nicht erkannt

epictechlab, 2021-01-15:
Hallo, 

Danke erstmal für das tolle Projekt.
Ich habe bereits die vorherige Version mit einer nodemcu am laufen gehabt, und will mir jetzt die neue Version auf Basis des Lolin32 aufbauen. Habe mir hierzu deine PCBs und die anderen Komponenten bestellt. Das einzige das ich nicht habe ist die Batterie, da ich den Tonuino nicht portable benötige.
Allerdings wird hier die SD Karte nicht erkannt und somit startet der Tonuino nicht in den AP mode. 
Kann es sein das ich die Batterie benötige auch wenn ich sie nicht brauche?
Die CHG LED des Lolin32 flackert auch immer wie wild.
Und mein neopixel blinkt rot. Was darauf schließt das die SD Karte nicht erkannt wird.

Woran könnte es denn noch liegen?

biologist79, 2021-01-15:
Moin,

steht auf deinem Lolin32 v1.0.0 drauf? Wenn ja: Dann ist das mit dem ständigen Blinken normal. Mir ist auch unklar, was das soll, aber es geht auf jeden Fall weg, wenn man das Board im Batteriebetrieb hat. Einen Akku brauchst du nicht. Schau dann halt, dass du die Spannungsüberwachung nicht mit einkompilierst.

Mit SD ist das so eine Sache. Üblicherweise liegt das an schlechten Verbindungen auf dem Breadboard (welches man ja üblicherweise nutzt, wenn man das alles mal testen will). Hatten schon einige User Probleme mit; zuletzt zB hier: https://github.com/biologist79/Tonuino-ESP32-I2S/issues/51
Verifiziere nochmal, dass die richtigen GPIOs angeschlossen sind und versuche es vielleicht nochmal mit anderen Jumperwires.
Der Neopixel blinkt rot, weil der Bootvorgang nicht komplettiert wird. Ich hatte mal eine Weile einen Modus, dass man auch ohne SD booten konnte, aber es macht dauerhaft einfach keinen Sinn. 

VG.

biologist79, 2021-01-20:
Neuigkeiten?

epictechlab, 2021-01-20:
Ja und nein, das erste Problem war ein GND Problem am neopixel Ring. Dadurch wurde nicht weiß sondern rot als Farbe dargestellt. Heißt der Lolin32 ist normal gebootet und eigentlich würde du SD auch erkannt.

Das eigentliche Problem ist aber das die deepsleep Schaltung mit den 2 MOSFETs nicht richtig funktioniert. Wenn ich diese umgehe und direkt 3,3v auf die source vom P-kanal MOSFET lege, dann läuft alles.

Pin 17 schaltet nach einer gewissen Zeit die Spannung weg. Womit der N-kanal MOSFET nicht mehr durchschaltet und damit den P-kanal MOSFET lahmlegt. 

Warum? Keine Ahnung.

Was mir noch aufgefallen ist, kann es sein das die maximal Lautstärke sehr gering ist. Mir kommt es sehr leise vor. Ich werde aber heute Abend Mal einen anderen Lautsprecher dran hängen und nochmal testen.

biologist79, 2021-01-20:
Wie groß ist denn dein Vorwiderstand am Gate des IRF530N? Ich hatte da auch mal 4.7k, aber es hat nicht funktioniert. Wenn ich damit auf 1k runtergegangen bin, hat es geklappt.

biologist79, 2021-01-20:
Achso, dass der Tonuino startet sich nach ein paar Sekunden automatisch neu, wenn das Flag `SHUTDOWN_IF_SD_BOOT_FAILS` gesetzt ist. Grund dafür ist, dass man im Batteriebetrieb, wenn was mit SD schiefläuft, sonst keine Möglichkeit mehr hat, den Tonuino neu zu starten (wenn alles eingebaut ist).

epictechlab, 2021-01-20:
> Wie groß ist denn dein Vorwiderstand am Gate des IRF530N? Ich hatte da auch mal 4.7k, aber es hat nicht funktioniert. Wenn ich damit auf 1k runtergegangen bin, hat es geklappt.

1k so wie es in deiner Anleitung angegeben ist.

epictechlab, 2021-01-20:
> Achso, dass der Tonuino startet sich nach ein paar Sekunden automatisch neu, wenn das Flag `SHUTDOWN_IF_SD_BOOT_FAILS` gesetzt ist. Grund dafür ist, dass man im Batteriebetrieb, wenn was mit SD schiefläuft, sonst keine Möglichkeit mehr hat, den Tonuino neu zu starten (wenn alles eingebaut ist).

Gut zu wissen, die kann ich auskommentieren da ich keine Batterie nutze.

biologist79, 2021-02-01:
Hast du das gelöst bekommen?
Hinweis an dieser Stelle: der IRF530N besitzt eine Vgs von 2 bis 4 V. Ich denke wenn man da ein Modell erwischt, das eher nach oben streut, dann schaltet der nicht durch. Man kann es einfach testen, indem man an das Gate des NDP6020p GND legt (Achtung: Gate ist NICHT der Anschluss in der Mitte!). Inzwischen verwende ich IRL3103, da dieser eine erheblich niedrigere Vgs besitzt.

closed 2021-02-08 by biologist79

epictechlab, 2023-08-13:
Hallo,
ist zwar nicht ganz sauber hier rein zuschreiben da das Thema "closed" ist, aber ichg mache es jetzt einfach trotzdem. ich konnte mich jetzt wirklich lange nicht mehr mit dem Projekt auseinander setzten und habe mittlerweile gesehen das sich hier einiges getan hat. ich würde meinen Kindern zu liebe das ganze nochmal neu aufrollen, oder besser gesagt optimieren.
Weil wie damals beschreiben gab es immer weider mal Probleme...

Ich habe damals diese PCB von dir verwendet? **TONUINO LOLIN32 3.3V rev2** sind die noch brauchbar oder ist das inzwischen alles überholt?
Welcher ESP wird denn aktuell verwendet? 

Evtl. findest ja mal kurz zeit mir hier zu antworten, Danke dir schonmal.

beste Grüße

tueddy, 2023-08-13:
Kenne jetzt Dein Board nicht aber grundsätzlich wird der Lolin-32 unterstützt, egal ob Softwareversion "master" oder "DEV".
Falls Probleme erstelleeinen Eintrag im ESPuino-Forum: https://forum.espuino.de