biologist79/ESPuino #82: Problems with old MP3 Tags (ID3v2.2)

vrjpellenz, 2021-02-01:
With the newest version I experienced problems with MP3 files with older MP3 Tags (ID3v2.2). All the songs with old tags did not play, and on the console the following messages appeared multiple times:
```
...
info        : ID3 framesSize=2337
info        : ID3 version=2
info        : ID3 version must be 3, skip all metadata
info        : ID3 framesSize=2337
info        : ID3 version=2
info        : ID3 version must be 3, skip all metadata
...
```
After writing the tag info with some MP3 Tag software, it was updated (as a side effect) to version ID3v2.3 and the files played again. Maybe a change in the underlying MP3 lib caused this new behavior. (It was working before with the older version of ESPuino / ESP TonUINO from December 2020.)

biologist79, 2021-02-01:
What's the version of the mp3-lib you're using? Just asking as a few weeks ago a method was introduced that's not yet supported by the last esp32-Arduino-Framework. That's why in platformio.ini I stepped away from the newest version (in order to avoid compiling-problems) and added "https://github.com/schreibfaul1/ESP32-audioI2S.git#a816a19".

vrjpellenz, 2021-02-01:
I am using the current platformio.ini:

```
[common]
lib_deps_builtin =
        SPI
        Wire
lib_deps_external =
        https://github.com/schreibfaul1/ESP32-audioI2S.git#a816a19                      ; currently not master-version as commit a816a19 not compatible with latest stable ESP32-Arduino
        https://github.com/madhephaestus/ESP32Encoder.git
        https://github.com/knolleary/pubsubclient.git
        https://github.com/biologist79/ESP32FTPServer
        https://github.com/FastLED/FastLED.git#3.4.0
        ESP Async WebServer
        https://github.com/me-no-dev/AsyncTCP
        https://github.com/bblanchon/ArduinoJson.git
; https://github.com/pschatzmann/ESP32-A2DP.git
```

which results in the following dependency tree:

```
|-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <ESP32-audioI2S-master> 2.0.0+sha.a816a19
|   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <ESP32Encoder> 0.3.8+sha.db7bbf2
|-- <PubSubClient> 2.8.0+sha.2d228f2
|-- <ESP32FTPServer> 0.0.1+sha.f71e97d
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <SPI> 1.0
|-- <FastLED> 3.4.0+sha.2fe6aeb
|   |-- <SPI> 1.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.17.2+sha.1360b6a
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <ArduinoJson> 6.17.2+sha.1360b6a
|-- <MFRC522> 1.4.6+sha.cd626f4
|   |-- <SPI> 1.0
|-- <PN5180 Library> 1.1.0+sha.2e3289f
|   |-- <SPI> 1.0
|-- <FS> 1.0
|-- <SD(esp32)> 1.0.5
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <SD_MMC> 1.0
|   |-- <FS> 1.0
|-- <AsyncTCP> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <nvsdump>
```

biologist79, 2021-02-08:
Have you been trying to use the recent version of the audio-lib (https://github.com/schreibfaul1/ESP32-audioI2S) that is used by ESPuino? A few days a ago there was a commit regarding ID3-stuff: https://github.com/schreibfaul1/ESP32-audioI2S/commit/29806a189cc0ecbfddcd61681b4a793d40e4c82b

tueddy, 2021-02-13:
Some MP3 played fine before but now i get the same error messages.
I have created an issue in the used audio lib:
https://github.com/schreibfaul1/ESP32-audioI2S/issues/113

tueddy, 2021-02-13:
The author of the library has fixed the issue in lightspeed.
Problem was reading ID3 tags in version 2.2

I think this issue can be closed..

closed 2021-02-17 by biologist79