biologist79/ESPuino #97: 16MB Speicher bei Lolin D32 pro nutzen.

gmbo, 2021-03-15:
Um den D32 pro, der häufig 16MB Flash besitzt zu nutzen müsste in die Platformio.ini 
board_upload.maximum_size = 16777216
board_upload.flash_size = 16MB
eingesetzt werden.
https://github.com/platformio/platform-espressif32/issues/482#issuecomment-773196654

biologist79, 2021-03-15:
Ja ich kann es mal aufnehmen. Allerdings ist das Image momentan auch nur knapp über 2 MB groß.

gmbo, 2021-03-15:
wenn aber jemand z.B.NVS vergrößert kann er es nutzen.
Da war ja zum Beispiel die Rede von App auf 2MB verkleinern. 
Es wäre dann sogar 4MB App möglich mit zusätzlich einiges an FAT oder SPIFFS

biologist79, 2021-03-15:
Jaja, mir sind die Vorteile schon klar.
Ich weiß nur noch gar nicht so genau, wo die Reise hingehen wird. Also grundsätzlich versuche ich derweil, dass man alles auch mit einem WROOM nutzen kann. Ob das immer so bleiben wird, das weiß ich noch nicht.

Eija, ich kann es bei Gelegenheit ja mal aufnehmen. Frisst ja kein Heu. Unpraktisch finde ich nur, dass man dne Chips nicht ansieht, wieviel PSRAM sie haben. Ich glaube, dass ich damals den Lolin D32 pro mit 16 Mb bestellt habe, aber so ganz sicher bin ich mir gar nicht.

Was mir aber eben bei einem Blick auf den ESP32 noch auffiel: Es steckt sogar ein WROVER-E drin. Also die aktuelle Revision.

biologist79, 2021-03-20:
Erledigt.

closed 2021-03-20 by biologist79