biologist79/ESPuino #100: Headphone PCB BOM Issue

dmlr1996, 2021-03-23:
Hello there,

some weeks ago, you have provided the new and detailed BOM for the PCB's. Now we have the problem, that in the BOM for the Headphone PCB is a IC1 (ME6211C33m5g) written but we cant find the IC1 in the Schematic / on the ordered PCB. Do you have a solution fot that?

biologist79, 2021-03-23:
Hi,

I forwarded this question to the guy that has been designing this PCB. He knows best, hehe.

compactflash, 2021-03-23:
Hi @dmlr1996 
Maybe you have an old version of the BOM , the current ist [here](https://u.pcloud.link/publink/show?code=kZjJVKkZOYso9U99qILNOMm5ehliaFtxldWX) . I have made a 5V-version too , but the released one is for 3.3V . The ME6211 is not required .

dmlr1996, 2021-03-23:
Hi @compactflash
thanks for your message. I have the BOM you provided to me in #93. I will check it again. Thanks.

dmlr1996, 2021-03-23:
I have checked the BOM in the link you provided. There is a IC1 written for the headphone PCB. It's the same version like I have already. 
[image]

dmlr1996, 2021-03-23:
Another thing we discovered today, neither the TDA1308 not the PT2308 are available at any of the nameable electronics distributors. Do you have any idea, how we can handle this?

biologist79, 2021-03-23:
At least something I can answer :-)
https://de.aliexpress.com/item/33020207730.html

dmlr1996, 2021-03-23:
Thanks @biologist79 the problem about it is the delivery time of 30-60 days .. xD

biologist79, 2021-03-23:
In case you need some and you're from Germany: 20 of those are sitting next to me :-)

dmlr1996, 2021-03-23:
Ja wenn das so ist, können wir auch deutsch schreiben :P
Das klingt gut. Mein kollege und ich bereiten gerade 5 Platinen vor, würdest du uns 5 Stück abgeben? Wenn ja, was sind wir dir schuldig?

biologist79, 2021-03-23:
Mach's so: Melde mich in meinem ESPuino-Forum (https://forum.espuino.de/) am besten mal an - das geht auch mit den Github-Zugangsdaten. Und schreibe mir dort ne PM. Dann machen wir das dort :-)

closed 2021-04-18 by dmlr1996