biologist79/ESPuino #108: LPCD with PN518 does not work

biologist79, 2021-07-27:
Needs to be investigated

tueddy, 2021-07-27:
Ja kann ich nachvollziehen. Habe den neuen Refactoring Code jetzt am Start und dort funktioniert LPCD nicht.
ESPuino wacht zwar auf aber friert dann ein.
Ich teste gerade mit meinem LPCD Demo, dort scheint es das gleiche Fehlerbild zu geben. Muss also nicht zwangsläufig eine Regression aus dem Refactoring sein. Checke das und melde mich sobald Ursache gefunden..

biologist79, 2021-07-27:
Dafür wäre ich dir sehr dankbar 👍

tueddy, 2021-07-27:
[Demo ](https://github.com/tueddy/PN5180-Library/commit/3a160558c13732eb4b93ec281a5305f33d2bd054)für ESP32 läuft wieder. Es gibt jetzt noch ein Problem im ESPuino refactoring code..

tueddy, 2021-07-28:
In der setup(); Routine fehlt Rfid_Init();, dadurch werden die eingefrorenen Eingänge nicht zurückgesetzt und der ESP bleibt komplett hängen;

` 
void setup() {

    Log_Init();

    Rfid_Init();  <-- Dieser Aufruf fehlt

    Rfid_WakeupCheck();

    System_Init();

`

Außerdem fehlt in Rfid_Init() die Initialisierung des IRQ PIN:

`   
void Rfid_Init(void) {

       #ifdef PN5180_ENABLE_LPCD

            // disable pin hold from deep sleep

            gpio_deep_sleep_hold_dis();

            gpio_hold_dis(gpio_num_t(RFID_CS));  // NSS

            gpio_hold_dis(gpio_num_t(RFID_RST)); // RST

            pinMode(RFID_IRQ, INPUT);  <-- Dieser Aufruf fehlt

        #endif

    }

`
Danach läuft LPCD wieder wie vor dem Refactoring.
Kannst Du die 2 Zeilen einpflegen? Ein PR ist hier zu aufwendig..
Schöne Grüße
tueddy

biologist79, 2021-07-28:
Habe es eingecheckt. Vielen Dank für deine Mühen!

closed 2021-07-28 by biologist79