biologist79/ESPuino #109: Geht das auch auf einen M5 Core ?

ChR-iSz, 2021-08-30:
Hi,

ich nutze ein M5 Core (Geniales Teil!).
Meinst du es ist möglich, den Code darauf laufen zu lassen, und den internen Speakter + SDCard Slot + Display anzusteuern ? Quasi -> All in one Lösung ?!

Viele Grüße...

biologist79, 2021-08-30:
Moin,

um ehrlich zu sein kann ich dir die Frage nicht beantworten.
Sie kam aber in meinem Forum bereits auf: https://forum.espuino.de/t/espuino-auf-dem-m5stack-gray-interner-dac/78
Also im Kern startet die Frage mit: Kriegst du die [Lib](https://github.com/schreibfaul1/ESP32-audioI2S), die ich zum Dekodieren verwende, darauf zum Laufen?

biologist79, 2021-08-30:
Führt das Teil eigentlich auch GPIOs nach außen? Also SD scheint via SPI angebunden zu sein, das kann man 1:1 nutzen. Aber die Frage ist, wo man den RFID-Reader anschließt. Wobei man notfalls auf einen RC522 mit i2c zurückgreifen könnte, weil i2c wird ja verwendet. Kommt man an die Pins gut dran?

schreibfaul1, 2021-08-30:
Hallo, ich vermute, dass es  funktionieren wird, testen kann ich das mangels HW nicht. In meinem Repository gibt es im Ordner examples Beispielcode für dem M5, der mir von den Usern zugeschickt wurde.

biologist79, 2021-08-30:
Danke @schreibfaul1 für deinen Kommentar. Hatte ich nicht auf dem Schirm mit diesen Beispielen.
Wenn es das schon wirklich ist, dann wird es vermutlich tatsächlich funktionieren.
Die Frage ist im Endeffekt dann nur, wo man den RFID-Reader anschließt.

schreibfaul1, 2021-08-30:
das könnte auch helfen:
https://macsbug.wordpress.com/2021/02/14/web_radio_of_m5stack_i2s_dac/

ChR-iSz, 2021-08-31:
Danke Euch beiden. Ich werde mich da heute Abend mal ran machen. In Arduino bzw. Platformio bindet man einfach die M5Stack Bibliothek ein, und hat Zugriff auf RFID, SDCARD, Display, etc.

Ja, die PINS sind alle nach außen geführt. Ich habe noch das Batterie Modul-dran (auf dem Foto rotes Modul). Damit bleibt der ESP ca. 2 Tage autark vom Strom.

Komponenten gibt es on Mass: https://eckstein-shop.de/M5Stack-EN

Der RFID wird einfach eingesteckt und kostet ein paar Euro bei Eckstein.

Was mir daran gefällt, ist u.a.
A: Cover und Hörspielanzeige auf dem Display (?) (evtl. kann man dann auch die Lieder mal angenehm auswählen).
B: Sehr kompaktes Gerät,
C: Modular Plug'n Play Sensoren.,
D: Ein Lautsprecher ist bereits verbaut. Evtl reicht der für Hörspiele vollkommen aus (?). Ein externen Lautsprecher kann einfach an PIN 26 + Ground (DAC) angeschloßen werden.

Ich nutze Platformio (Arduino) in Visual Studio Code. Wenn ich soweit bin, werde ich Euch mal berichten, ob das alles so hingehauen hat.

Vielen dank nochmal für die schnelle Unterstützung ! 👍

[image]

[image]

biologist79, 2021-08-31:
Klingt auf jeden Fall nach einem Projekt, das du in meinem Forum beschreiben könntest :-)

biologist79, 2021-12-02:
Gibt's hier Neuigkeiten oder kann das geschlossen werden?

ChR-iSz, 2021-12-05:
Ich denke das kann geschlossen werden. Ich habe es zwar geschafft, aber der interne Lautsprecher hat ein deutliches knacksen/rauschen beim abspielen.

VG

closed 2021-12-06 by biologist79

drtrigon, 2023-09-19:
> Hallo, ich vermute, dass es funktionieren wird, testen kann ich das mangels HW nicht. In meinem Repository gibt es im Ordner examples Beispielcode für dem M5, der mir von den Usern zugeschickt wurde.

Wo ist das Beispiel? Ich würde ESPuino gerne auf einem M5Core2 mit RCA Module und M5GO Bottom 2 (RGB LEDs, kindgerechter Ladeanschluss) testen.

biologist79, 2023-09-19:
Wir hatten das mal andiskutiert, aber ich wüsste nicht, dass das mal jmd. fertiggebracht hätte:
https://forum.espuino.de/t/espuino-auf-dem-m5stack-gray-interner-dac/78