biologist79/ESPuino #119: MQTT: new values not published

GeneralLuzi, 2021-11-21:
Hi,

while setting a new value for the loudness is working fine, brightness does not work properly IMHO.

Use Case for loudness:
1) ESPUINO_MQTT publishes on start e.g. 10.
2) PRIVATE_MQTT_CLIENT receives 10 
3) PRIVATE_MQTT_CLIENT publishes 11
4) ESPUINO_MQTT receives 11 and espuino sets loudness to 11
5) ESPUINO_MQTT  publishes 11
6) PRIVATE_MQTT_CLIENT receives 11 

Use Case for brightness:
1) ESPUINO_MQTT publishes on start e.g. 10.
2) PRIVATE_MQTT_CLIENT receives 10 
3) PRIVATE_MQTT_CLIENT publishes 11
4) ESPUINO_MQTT receives 11 and espuino sets brightness to 11
**-- NOT WORKING BELOW THIS LINE --**
5) ESPUINO_MQTT  publishes 11
6) PRIVATE_MQTT_CLIENT receives 11

Thanks and regards,
GL

GeneralLuzi, 2021-11-21:
Same on:
- topicLockControlsCmnd
- Cmnd/ESPuino/SleepTimer

biologist79, 2021-11-21:
Need to investigate.

thebino, 2021-11-22:
I discovered the same issue today. Fix can be found here: https://github.com/biologist79/ESPuino/pull/124

GeneralLuzi, 2021-11-22:
If you need a tester feel free to ask! :)Von: Benjamin Stürmer ***@***.***>Gesendet: Montag, 22. November 2021, 18:38An: biologist79/ESPuinoCc: GeneralLuzi; AuthorBetreff: Re: [biologist79/ESPuino] MQTT: new brightness values not published (Issue #119)
I discovered the same issue today. Fix can be found here: #124

thebino, 2021-11-22:
@GeneralLuzi You can just checkout the branch and deploy it to your device. It will be reviewed this week anyway :D

GeneralLuzi, 2021-11-22:
@thebino my test shows 2 out of 3. Well done!  👍 

Brightness and SleepTimer I can confirm to be fixed. Somehow LockControls is not publishing the change! Could you have a look please?

Thanks!

thebino, 2021-11-22:
@GeneralLuzi can you pull the latest state and validate if it updates the state now? Haven't tested myself. Just added the fix and pushed it.

biologist79, 2021-11-23:
Merged into master.
@GeneralLuzi Please review if it's working for you.

GeneralLuzi, 2021-11-23:
Thanks @biologist79 

I retested the feature on the branch of @thebino this morning and it worked. Afterwards I wanted to test the RFID commands in combination with MQTT and discovered a major fail - my SD Card (filesystem?!) was destroyed... Not sure why and how... I am trying to recap... maybe a new issue - maybe just bad luck with the SD card not sure

After fixing the SD Card, I will test the merge and comment again...

biologist79, 2021-12-02:
Can this one be closed?

GeneralLuzi, 2021-12-03:
I will close this issue. Thanks for fixing. If there is anything left, I will re-open again.

closed 2021-12-03 by GeneralLuzi