biologist79/ESPuino #120: Enabled HEADPHONE_ADJUST_ENABLE: RFID not working

GeneralLuzi, 2021-11-21:
Hi,

having only three modules activated:
- LANGUAGE DE  
- HEADPHONE_ADJUST_ENABLE
- SHUTDOWN_IF_SD_BOOT_FAILS

Issues:
1. RFID (PN5180) is not working
2. log console every second: ws[/ws][1] text-message[24]: ping

Deactivating HEADPHONE_ADJUST_ENABLE --> RFID is working + no strange log console

Thanks and best regards,
GL

GeneralLuzi, 2021-11-21:
by the way, my headphone pcb will arrive tomorrow - so it is not connected. Do not know, if a missing headphone pcb could trigger this issue...

biologist79, 2021-11-21:
Sounds like overlapping GPIOs in your settings-files.

biologist79, 2021-11-21:
ping is coming from a webbrowser with the webgui opened. Just close it and it's gone.

GeneralLuzi, 2021-11-21:
my bad: thanks for the investigating.

RFID RST GPIO PIN was set to HP_DETECT GPIO PIN

closed 2021-11-21 by GeneralLuzi

biologist79, 2021-11-21:
Is a common mistake - don't worry :-)