biologist79/ESPuino #121: WebGUI: stopped track / playlist - current title is set

GeneralLuzi, 2021-11-21:
Hi,

Use case:
1. play song
2. stop song via MQTT: Cmnd/ESPuino/TrackControl --> 1
    **Issue here: ws[/ws][1] text-message[24]: ping --> every 15 seconds**
3. go on website to Controll (Steuerung)
    **Issue here: current title (Aktuelller titel) is set --> should be blank IMHO**
4. trying to resume does not start the song --> console: Playmode kann nicht verändert werden, wenn keine Playlist aktiv ist. --> IMHO correct behaviour

biologist79, 2021-11-21:
Ping is no issue. It indicates active conversation between uC and browser.
Regarding STOP: it's probably not tested well as usually only play/pause is used. Need to have a look.

biologist79, 2021-11-28:
Fixed point3. 
Please test.

GeneralLuzi, 2021-11-28:
works great! Thx.

closed 2021-11-28 by GeneralLuzi