biologist79/ESPuino #125: D32 pro: build in sd card reader in coflict with PN5180

GeneralLuzi, 2021-11-28:
Hi,

it seems, that the build in sd card reader of the D32 pro is not compatible with PN5180-NFC. Let us see if it is a layer 8 issue... ;)

Unneeded pins are set to 99.

The following scenarios are tested:
1. MFRC522_SPI activted, PN5180 deactivated, RFID PINS to 99 --> OK
2. MFRC522_SPI deactivted, PN5180 activated, RFID PINS to 99 --> NOK
3. MFRC522_SPI deactivted, PN5180 activated + PINs set to default, but PN5180 not connected --> NOK
4. MFRC522_SPI deactivted, PN5180 activated + PINs set to default, PN5180  connected --> NOK

(Not tested:  MFRC522_SPI activted, PN5180 deactivated, RFID PINS to default --> I do not have a MFRC522)

Definition of OK: web server is accessible and music can be played. 

Why NOK: Output is constant:

```  
Maximale Inaktivitätszeit wurde aus NVS geladen: 10
Firmware version=0.0
RFID-Tags koennen jetzt gescannt werden...
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x37
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x29
[E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
SD-Karte konnte nicht gemountet werden.
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
[E][sd_diskio.cpp:775] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[E][sd_diskio.cpp:194] sdCommand(): Card Failed! cmd: 0x00
SD-Karte konnte nicht gemountet werden.
```  

SW-Version:
```
ESP-IDF-version (major): 3
ESP-IDF-version (minor): 3
Software-revision: 20211123-1
```

settings.h:

```  

    //########################## MODULES #################################
    //#define PORT_EXPANDER_ENABLE          // When enabled, buttons can be connected via port-expander PCA9555 (https://forum.espuino.de/t/einsatz-des-port-expanders-pca9555/306)
    //#define I2S_COMM_FMT_LSB_ENABLE       // Enables FMT instead of MSB for I2S-communication-format. Used e.g. by PT2811. Don't enable for MAX98357a, AC101 or PCM5102A)
    #define MDNS_ENABLE                     // When enabled, you don't have to handle with ESPuino's IP-address. If hostname is set to "ESPuino", you can reach it via ESPuino.local
    //#define MQTT_ENABLE                     // Make sure to configure mqtt-server and (optionally) username+pwd
    //#define FTP_ENABLE                      // Enables FTP-server; DON'T FORGET TO ACTIVATE AFTER BOOT BY PRESSING PAUSE + NEXT-BUTTONS (IN PARALLEL)!
    //#define NEOPIXEL_ENABLE                 // Don't forget configuration of NUM_LEDS if enabled
    //#define NEOPIXEL_REVERSE_ROTATION     // Some Neopixels are adressed/soldered counter-clockwise. This can be configured here.
    #define LANGUAGE DE                     // DE = deutsch; EN = english
    //#define STATIC_IP_ENABLE              // Enables static IP-configuration (change static ip-section accordingly)
    //#define HEADPHONE_ADJUST_ENABLE       // Used to adjust (lower) volume for optional headphone-pcb (refer maxVolumeSpeaker / maxVolumeHeadphone) and to enable stereo (if PLAY_MONO_SPEAKER is set)
    //#define PLAY_MONO_SPEAKER               // If only one speaker is used enabling mono should make sense. Please note: headphones is always stereo (if HEADPHONE_ADJUST_ENABLE is active)
    #define SHUTDOWN_IF_SD_BOOT_FAILS       // Will put ESP to deepsleep if boot fails due to SD. Really recommend this if there's in battery-mode no other way to restart ESP! Interval adjustable via deepsleepTimeAfterBootFails.
    //#define MEASURE_BATTERY_VOLTAGE         // Enables battery-measurement via GPIO (ADC) and voltage-divider
    //#define PLAY_LAST_RFID_AFTER_REBOOT   // When restarting ESPuino, the last RFID that was active before, is recalled and played
    //#define USE_LAST_VOLUME_AFTER_REBOOT  // Remembers the volume used at last shutdown after reboot
    //#define USEROTARY_ENABLE                // If rotary-encoder is used (don't forget to review WAKEUP_BUTTON if you disable this feature!)
    //#define BLUETOOTH_ENABLE                // If enabled and bluetooth-mode is active, you can stream to your ESPuino via bluetooth (a2dp-sink).
    //#define IR_CONTROL_ENABLE             // Enables remote control (https://forum.espuino.de/t/neues-feature-fernsteuerung-per-infrarot-fernbedienung/265)
    //#define CACHED_PLAYLIST_ENABLE          // Enables playlist-caching (infos: https://forum.espuino.de/t/neues-feature-cached-playlist/515)
    //#define PAUSE_WHEN_RFID_REMOVED       // Playback starts when card is applied and pauses automatically, when card is removed (https://forum.espuino.de/t/neues-feature-pausieren-wenn-rfid-karte-entfernt-wurde/541)
    //#define SAVE_PLAYPOS_BEFORE_SHUTDOWN  // When playback is active and mode audiobook was selected, last play-position is saved automatically when shutdown is initiated
    //#define SAVE_PLAYPOS_WHEN_RFID_CHANGE // When playback is active and mode audiobook was selected, last play-position is saved automatically for old playlist when new RFID-tag is applied

    //################## select SD card mode #############################
    //#define SD_MMC_1BIT_MODE              // run SD card in SD-MMC 1Bit mode (using GPIOs 15 + 14 + 2 is mandatory!)
[... 125 lines cut ...]
    #endif

    // (Optional) remote control via infrared
    #ifdef IR_CONTROL_ENABLE
        #define IRLED_PIN                   99//22              // GPIO where IR-receiver is connected (only tested with VS1838B)
        #define IR_DEBOUNCE                 200             // Interval in ms to wait at least for next signal (not used for actions volume up/down)

        // Actions available. Use your own remote control and have a look at the console for "Command=0x??". E.g. "Protocol=NEC Address=0x17F Command=0x68 Repeat gap=39750us"
        // Make sure to define a hex-code not more than once as this will lead to a compile-error
        // https://forum.espuino.de/t/neues-feature-fernsteuerung-per-infrarot-fernbedienung/265
        #define RC_PLAY                     0x68            // command for play
        #define RC_PAUSE                    0x67            // command for pause
        #define RC_NEXT                     0x6b            // command for next track of playlist
        #define RC_PREVIOUS                 0x6a            // command for previous track of playlist
        #define RC_FIRST                    0x6c            // command for first track of playlist
        #define RC_LAST                     0x6d            // command for last track of playlist
        #define RC_VOL_UP                   0x1a            // Command for volume up (one step)
        #define RC_VOL_DOWN                 0x1b            // Command for volume down (one step)
        #define RC_MUTE                     0x1c            // Command to mute ESPuino
        #define RC_SHUTDOWN                 0x2a            // Command for deepsleep
        #define RC_BLUETOOTH                0x72            // Command to enable/disable bluetooth
        #define RC_FTP                      0x65            // Command to enable FTP-server
    #endif
#endif
```

biologist79, 2021-11-28:
In general SPI-bus can be shared by multiple devices. If doing so, all SPI-devices share MOSI/MISO/SCK, but every SPI-devices needs a dedicated CS (chip select). If SPI should be shared this can be enabled for ESPuino via "SINGLE_SPI_ENABLE". However, I never got this working reliable with SD-SPI + RC522. And I never tested it for PN5180 + SD-SPI.
Unfortunately there's a bug with SD-SPI: when doing uploads sometimes because of and internal error (guess it's an ESP32-Arduino-problem) ESP32 stops writing to SD-card. Have not been able to solve this so far.

I'd strongly suggest to use SD_MMC instead of SD-SPI because...
- SD_MMC only needs 3 PINs
- SD_MMC is faster
- SD_MMC does not show this upload-problem

To address this, I developed a PCB that makes use of Lolin D32 pro but with an external SD-Module: https://forum.espuino.de/t/lolin-d32-pro-mit-sd-mmc-pn5180-max-fuenf-buttons-und-port-expander-smd/638

GeneralLuzi, 2021-11-28:
I tested "SINGLE_SPI_ENABLE" on with the PN5180. 

The behavior is strange: The SD card issue is gone - however, any rfid card is recognized.  Also I double checked the pin allocation... seems not to work...

GeneralLuzi, 2021-11-28:
also the sound is not working...

GeneralLuzi, 2021-12-03:
I propose to update the HAL for "settings-lolin_d32_pro.h" and refer to this behavior. I was not fully aware of that issue with the build-in sd card.

Now having an external sd card reader and a d32 pro it greatly works... Having had this within the comments beforehand it might saved a lot of time...

Thanks for your hints (and of course your time to reply!)

Please feel free to close this issue.

biologist79, 2021-12-03:
This is already covered:
https://github.com/biologist79/ESPuino/blob/master/src/settings-lolin_d32_pro.h#L22

closed 2021-12-03 by biologist79