biologist79/ESPuino #126: Sleep Mode: Deactivating EOP, EOT, and EO5T not possible

GeneralLuzi, 2021-11-28:
Hey,

```
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: EOT]
Sleep-Timer: Nach dem Ende des laufenden Tracks.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 0]
Sleep-Timer ist bereits deaktiviert.
```

After  EOP, EOT, and EO5T is set, it cannot be deactivated by mqtt command. The device device goes to deep sleep mode although the log says the timer is deactivated.

Thanks and best regards

GL

biologist79, 2021-11-28:
Thanks for your report!
Just provided a fix - please test.

GeneralLuzi, 2021-11-28:
nope - sorry. 

The deactivation is not published via mqtt.

The log also does not indicate the fix:
```
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: EOT]
Sleep-Timer: Nach dem Ende des laufenden Tracks.
ws[/ws][1] text-message[24]: ping
RSSI: -75 dBm
MQTT: Bin noch online.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 0]
Sleep-Timer ist bereits deaktiviert.
```

GeneralLuzi, 2021-11-28:
also the device went into deep sleep mode

```
ESP-IDF-version (major): 3
ESP-IDF-version (minor): 3
Software-revision: 20211128-2
```

biologist79, 2021-12-01:
Just reviewed the code yesterday and discovered the problem. I'm looking forward to fix it this evening.

biologist79, 2021-12-02:
2nd attempt :-)
Please test.

GeneralLuzi, 2021-12-03:
Hi,

seems to work:
```
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: EOP]
Sleep-Timer: Nach dem letzten Track der Playlist.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 0]
Sleep-Timer wurde deaktiviert.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: EOT]
Sleep-Timer: Nach dem Ende des laufenden Tracks.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 0]
Sleep-Timer wurde deaktiviert.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: EO5T]
Sleep Timer: Nach Ende des Titels oder, wenn früher, Ende der Playlist
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 0]
Sleep-Timer wurde deaktiviert.
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 1]
LEDs wurden auf Nachtmodus gedimmt.
Modifikator: Sleep-Timer per RFID aktiviert (2 Stunden).
Sleep-Timer gesetzt auf: 1 Minute(n)
MQTT-Nachricht empfangen: [Topic: Cmnd/ESPuino/SleepTimer] [Command: 0]
Sleep-Timer wurde deaktiviert.
```
Also MQTT of the commands are correctly published! Well done!

However, I guess there is still an inconsistent behavior:
1. While SleepTimer 1 Min reduces the brightness of the LEDs, the other commands don't.
2. The deactivation of the SleepTimer should brighten up the LED, shouldn't it?

Thanks for your work so far!

biologist79, 2021-12-03:
Done. By doing so I also discovered there was no dimming enabled at EOP, EOT and EO5T. 
Fixed this inline.

Please test.

GeneralLuzi, 2021-12-03:
Positiv first:
EOT EO5T EOP decrease the lights, disabling sleep mode increases the light. So far so good.

However, there is no MQTT publish of the new brightness level. Could you please observe?

Thanks!

biologist79, 2021-12-03:
Tried it again :-)

GeneralLuzi, 2021-12-05:
Almost. :D

The following scenario does not work:
1. Play song with Brightness 16
2. configure EOT,  EO5T or something else 
3. de- or increase Brightness 
4. deactivate sleep timer --> Issue: brightness resets to 16 and not to the value change in between.

biologist79, 2022-12-28:
Finally fixed it after taking almost forever 😂.

closed 2022-12-28 by biologist79