biologist79/ESPuino #128: Reboot after cleaned sd card

GeneralLuzi, 2021-12-03:
Hi,

this is for sure a minor bug...

scenario:
1. save music on SD card and bind it to an RFID card
2. erase SD card --> I guess that a deletion of a single file instead of the whole SD card is also within this scenario
3. try to play music (non-existing) via RFID Card

Behavior:
- LEDs flashing wild and blue
- esp32 reboots  

Desired behaviour:
- unbind connection playlist and RFID ID, if RFID ID is bind and file is not found
- No reboot 

Logs:
```
RFID-Karte empfangen: 000023146130
[E][vfs_api.cpp:64] open(): /sdcard/musicfox_its_not_easy.mp3 does not exist
Datei oder Verzeichnis existiert nicht 
Fehler aufgetreten!
Kontroll-Kommando empfangen via Queue: 1
Playmode kann nicht verändert werden, wenn keine Playlist aktiv ist.
ws[/ws][1] connect
```

Thanks for reply.

biologist79, 2021-12-03:
I do not agree with automatic unbind. IMHO it's sufficient to "catch" it.
But yes: we don't have two opinions in regard that restart shouldn't occur in this case.
Guess this was introduced by accident with a recent change as this has already been working in the past. Will fix it. 
Thanks for your report.

biologist79, 2022-01-31:
Fixed with the latest commit. Thanks for your report.
Please test.

commit 4d6e7e55 (2022-01-31): Fix #128
closed 2022-01-31 by biologist79