biologist79/ESPuino #129: Button 3 without rotary

GeneralLuzi, 2021-12-03:
Hi,

if rotary is disabled, there is no chance to use button_3 aka ROTARYENCODER_BUTTON.

I suggest, to exclude ROTARYENCODER_BUTTON from the USEROTARY_ENABLE section and change the other code parts allowing ROTARYENCODER_BUTTON to be used as a normal button without having a rotary.

```
// Rotary encoder
    #ifdef USEROTARY_ENABLE
        #define ROTARYENCODER_CLK           34          // If you want to reverse encoder's direction, just switch GPIOs of CLK with DT (in software or hardware)
        #define ROTARYENCODER_DT            39          // 39 = 'VN'; rotary's encoder 
    #endif

    // Amp enable (optional)
    #define GPIO_PA_EN                      108         // To enable/disable amp for loudspeaker; connected to port-expander

    // Control-buttons
    #define NEXT_BUTTON                     102         // Next-Button: connected to port-expander
    #define PREVIOUS_BUTTON                 100         // Prev-Button: connected to port-expander
    #define PAUSEPLAY_BUTTON                101         // Pplay-Button: connected to port-expander
----------new---------
    #define ROTARYENCODER_BUTTON        103         // Set to 99 to disable the button; connected to port-expander
-------------------------
    #define BUTTON_4                        104         // Button 4: connected to port-expander
    #define BUTTON_5                        105         // Button 5: connected to port-expander
```

Thanks for consideration.

Best regrads,
GL

biologist79, 2021-12-03:
Technically you're right. But anyway I'm not sure if I really want to change this. Will give it a thought.

biologist79, 2022-01-31:
Changed according your wishes (finally, hehe).

commit 9e9af20c (2022-01-31): Fix #129 / Enable using button of RE without having RE enabled entirely
closed 2022-01-31 by biologist79