biologist79/ESPuino #130: USE_LAST_VOLUME_AFTER_REBOOT not working properly

GeneralLuzi, 2021-12-03:
Hi,

scenario 1:
- set volume to 11 without playing a song. In my case the starting point was 3. 
- go to deep sleep.
- wake up esp32 
- Issue: Volume is set to 9

scenario 2: 
- set volume to 11 without playing a song. In my case the starting point was 3. 
- go to deep sleep.
- reboot esp32 
- Issue: Volume is set to 9

Might be an issue.

Thanks and regards,
GL

anoff, 2022-01-24:
Is your max volume set to 9 by any chance? 👀

I also had issues with this feature in the most recent SW and decided to disable it for now. But in my case the volume always started ~3

biologist79, 2022-01-24:
Didn't forget you guys, sorry for the delay :-)
Will have a closer look ASAP. There's currently other stuff that needs my time.

biologist79, 2022-01-30:
Cannot reproduce this issue with the current master (from today).

biologist79, 2022-01-30:
Hmm, I cannot reproduce this issue with the current master.

anoff, 2022-02-06:
I will retry with the latest version in the coming days. Was busy programming cards so I didn't see your response, sorry :D

closed 2022-02-24 by biologist79