biologist79/ESPuino #176: Increase readability + Refactoring

bliepp, 2022-12-08:
Hello guys, nice project. While diving into the code for a feature I wanted to add I found some parts to be very hard to read (at least for me). I broke it down to two aspects:
* Inconsistency when it comes to formatting (e.g. indentation in Led.cpp/Led.h is 4 spaces while in Log.cpp - but not Log.h - it is 2 spaces) and naming conventions). Also sometimes functions are declared where you wouldn't expect them (e.g. in Led.h there are declarations for AudioPlayer_GetInitVolume and AudioPlayer_SetInitVolume - twice) and function/variable handling could be improved.
* Sometimes heavy nesting/very long function bodies (OMG Led_Task ^^)

So as a result I would suggest to 
1. Introduce a coding styleguide and enforce it with something like [EditorConfig](https://editorconfig.org/) and
1. While applying it step by step also refactor the code into a more readable form (e.g. replacing the _Namespace_FunctionName_ style and using C++ specific syntax like namespaces or classes, using guard clauses, etc.)

Kindly
bliepp

biologist79, 2022-12-08:
Hi,

feel free to refactor + test :-)
I'm already spending by far too much time for ESPuino. So I won't.

bliepp, 2022-12-08:
Okay, my wife's gonna kill me but I'll try my best. Do you prefer one big PR refactoring big chunks or many small PR's refactoring refactoring module by module?

biologist79, 2022-12-08:
Regarding different indentions: was probably introduced because there've been working different people on it. Usually I take care that it looks the same but this is probably not the case everywhere.

LED is hard to understand - I know. There's also an idea to test another Neopixel-lib. But as I said: I' not willing to spend additional time. I'm currently working on PCB-stuff. And I guess this won't change in near future.

Don't know what's the best way to move to C++. Probably a good startpoint is to start a conversation at ESPuino's forum. Maybe others are interested, too.

fschrempf, 2022-12-24:
I just want to confirm that at least the code in the LED Task desperately needs some refactoring. It's barely readable at the moment. Thanks @bliepp for stepping up! I highly appreciate the effort!

bliepp, 2022-12-24:
> I just want to confirm that at least the code in the LED Task desperately needs some refactoring. It's barely readable at the moment. Thanks @bliepp for stepping up! I highly appreciate the effort!

This is actually the first thing I want to refactor as it seems to be the biggest load of them all. I'm just waiting for PR #171 to be merged to avoid merge conflicts.

fschrempf, 2022-12-25:
> The LED implementation is actually the first thing I want to refactor as it seems to be the biggest load of them all. I'm just waiting for PR #171 to be merged to avoid introducing a conflict.

You might want to look at https://github.com/biologist79/ESPuino/pull/184 which includes refactoring and style fixes for the LED volume indicator code. It's only a small part of what needs to be done, but anyway. Feel free to let me know what you think about it.

bliepp, 2022-12-25:
> You might want to look at #184 which includes refactoring and style fixes for the LED volume indicator code. It's only a small part of what needs to be done, but anyway. Feel free to let me know what you think about it.

I just saw your PR and so far I think it looks great. As I don't want to mess with others' efforts I think I will refactor and restructure other parts of the software first until your draft is finished and I can safely mess with the LED part. I'm very thankful for any additional refactoring efforts.

fschrempf, 2022-12-25:
> I just saw your PR and so far I think it looks great. As I don't want to mess with others' efforts I think I will refactor and restructure other parts of the software first until your PR merge is finished and I can safely edit the LED portion. I'm very thankful for any additional refactoring efforts.

I already removed the draft status. In my opinion the PR is ready for reviewing and merging. Some tests by other users would be helpful of course. A big refactoring PR is probably not useful anyway. Taking things in small steps is probably best.

bliepp, 2022-12-25:
> A big refactoring PR is probably not useful anyway. Taking things in small steps is probably best.

While I would generally agree I don't think it is easy to restructure some parts in small steps. E.g. I plan on using C++ features like classes and namespaces and those require changes in the overall structure of some bits. E.g. encapsulating everything of the LED part into a class or namespace would hardly be possible without major changes.

fschrempf, 2022-12-25:
> While I would generally agree I don't think it is easy to restructure some parts in small steps. E.g. I plan on using C++ features like classes and namespaces and those require changes in the [...]

Sure, there is also a need for some treewide changes. That's fine. I was more thinking about the changes needed in the code flow/logic, like in #184, that also need some reviewing and testing. Reworking the logic of the LED code in one single PR is probably not the best option.

fschrempf, 2023-11-01:
One major part of this issue has been resolved in the current dev branch by introducing code formatting through `clang-format` including a CI linting job. See:  https://github.com/biologist79/ESPuino/pull/264.

There also have been several improvements of the code logic to increase the readability, especially in Led.cpp.

There is still room for improvements in this area, though. :wink:

tueddy, 2023-11-01:
LED refactoring + codestyle done ;-)

closed 2023-11-01 by tueddy