biologist79/ESPuino #179: Sporadic Artifacts in WS2812 LEDs

Joe91, 2022-12-18:
There are sporadic artefacts / pixel errors. They are more likely at the higher the pixel-indexs. 
More noticable when using more LEDs.

Best guess so far:
Maybe the transmission gets interrupted / delayed by some Ethernet IRQs or Flash access?

Any idea on how to further narrow this down or on what to test further welcome:-).

Thanks for this great project!

biologist79, 2022-12-18:
This is a known issue. Try to change LED-task's core from 0 to 1:
https://github.com/biologist79/ESPuino/blob/master/src/Led.cpp#L72

Joe91, 2022-12-19:
Sorry, I haven't found any post to this topic in the forum that's why I created this issue.

Thank you very much! This really solves the visual issues.
What are the downsides of this (why is this not the default)?

biologist79, 2022-12-19:
It's not default because sometimes "tiny stuff" can have massive side-effects @ESP32. As I currently don't have time to test it, it's not yet default.

closed 2022-12-19 by biologist79

Joe91, 2022-12-19:
Thanks a lot for this answer and your support!
Do you prefer to use the forum for such stuff?

biologist79, 2022-12-19:
Yes, I do :-)