biologist79/ESPuino #182: Bug: Audio for CMD_TELL_IP_ADDRESS disturbed

Joe91, 2022-12-19:
When using the CMD_TELL_IP_ADDRESS the speech is very hard to understand and completely messed up.
When pressing "pause" while the speech runs, it suddenly is understandable...
Attached a small recording of the difference.

[audio_speech.zip](https://github.com/biologist79/ESPuino/files/10262546/audio_speech.zip)

biologist79, 2022-12-19:
Has been working when I implemented it. However, there've been some issues with Google TTS because the number of requests was (almost certainly) somewhat too high so Google started to refuse those calls. So @Schreibfaul changed the provider and now it's broken.
However, I have to admit that I didn't test if it's a problem of the underlying lib or ESPuino.

Joe91, 2022-12-19:
Thanks for your response!
If you like we can continue on the forum.
I will have a look at the commit-history. Maybe I can find something there...

biologist79, 2022-12-25:
Fixed.

closed 2022-12-25 by biologist79