biologist79/ESPuino #203: RC522 does not process any rfid-codes

daywalker7911, 2023-02-28:
The log show that the rfid card is recognized, but the code is not transmittet to the rfid queue

laszloh, 2023-02-28:
Hi,
We'll need more information to find the bug. 

Can you post the following information:
- which target are you using 
- a diff of your settings.h (or a list of features which differ from the [default](https://github.com/biologist79/ESPuino/blob/master/src/settings.h))?
- which revision of ESPuino are you using (if you are not using the master/HEAD)
- steps you did (f.e. if the bug happened with a card in NVS or with an unknown card, or with no cards at all)
- Any other information which can help to reproduce the bug

Also an extract of the Log would be good.

biologist79, 2023-02-28:
Should be fixed now. Please test.
@laszloh You're totally right. But he already wrote me a private message.

biologist79, 2023-03-24:
Issue still ongoing?

closed 2023-04-04 by biologist79