biologist79/ESPuino #225: Forum offline

pitlochry, 2023-04-09:
Das Forum https://forum.espuino.de/ ist in den letzten Tagen und Wochen sehr häufig offline. Das soll keine Beschwerde sein, nur ein Hinweis, falls du es nicht weißt. Vielen Dank für deine Arbeit und das zur Verfügung stellen des Forums.

biologist79, 2023-04-09:
Der gesamte Server unterliegt zwei Monitorings und die zeigen nahezu 0 Downtime.
Ausfälle gab es nur, wenn ich Apache oder Docker aktualisiert habe. Aber das waren 2-3 Events in den letzten Monaten und insgsamt war das nicht mehr als fünf Minuten.

Kurzum: Ich kann das Problem nicht nachvollziehen.

pitlochry, 2023-04-09:
Danke für die schnelle Rückmeldung. Ich habe das Problem seit min. 10 Tagen (erst seit dem bin ich in die Thematik eingestiegen). Heute z.B. gegen 15 Uhr ging es für min. 1 Stunde nicht. Und an jedem der vorherigen Tage auch mehrere Stunden down. Gerade (20:30) geht er aber wieder.

biologist79, 2023-04-09:
Dann ist das ein Problem mit dem Routing.
Ping? Traceroute? Welchen Provider nutzt du?

pitlochry, 2023-04-09:
Provider: HSE Medianet GmbH 'ENTEGA Medianet GmbH'

I will ping and traceroute once again I can't get connection.

```
PING forum.espuino.de (78.46.91.78) 56(84) bytes of data.
64 bytes from ts-cs.de (78.46.91.78): icmp_seq=1 ttl=57 time=49.6 ms
```

```
traceroute to forum.espuino.de (78.46.91.78), 30 hops max, 60 byte packets
 1  _gateway (192.168.178.1)  18.913 ms  19.117 ms  19.087 ms
 2  lo0.ju7018.net-mn.de (213.157.2.18)  54.794 ms  58.939 ms  64.490 ms
 3  ae18.ju7001.net-mn.de (95.131.96.177)  52.046 ms  52.242 ms  58.555 ms
 4  ae1.ju7004.net-mn.de (95.131.96.173)  53.251 ms  58.356 ms  53.364 ms
 5  decix-gw.hetzner.com (80.81.192.164)  57.433 ms  53.679 ms  53.548 ms
 6  core23.fsn1.hetzner.com (213.239.224.65)  64.659 ms core24.fsn1.hetzner.com (213.239.224.81)  55.652 ms  55.090 ms
 7  ex9k1.dc3.fsn1.hetzner.com (213.239.229.254)  55.255 ms  48.854 ms  50.631 ms
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *
```
(do not know what these *** mean, hope it is save to post it)

biologist79, 2023-04-09:
Hmm, das sieht ok aus.
Zum traceroute: Das habe ich ausgeblockt fällt mir gerade ein.
Also sieht ok aus.
Kannst ja mal ins Forum schreiben, ob andere Leute Probleme haben. Mir sind keine bekannt.

pitlochry, 2023-04-09:
You were right, it is my fault. On my device I cannot connect at the moment, but on another device on the same network, it connects. Maybe, some resolve issue... Very strange, since this is the only website of myriads websites which I cannot connect to. And it happens arbitrarily.

```
ping forum.espuino.de
ping: forum.espuino.de: Name or service not known
```
```
traceroute forum.espuino.de
forum.espuino.de: Name or service not known
Cannot handle "host" cmdline arg `forum.espuino.de' on position 1 (argc 1)
```

closed 2023-04-09 by pitlochry