biologist79/ESPuino #308: Inconsistency in API definition

Captain-Sandwich, 2024-03-04:
https://github.com/biologist79/ESPuino/blob/1c8195673f2e676de582f1d76fd88139fe8ac88f/REST-API.yaml#L35C1-L37C64

The API definition above it specifies that a `GET` of `/explorer` should return an array of objects with `isDir = true` if a file is a directory.
Instead, the field is named `dir` and only returned for directories. I.e. there is no `dir=false` for files.

I think it can be fixed [here](https://github.com/biologist79/ESPuino/blob/1c8195673f2e676de582f1d76fd88139fe8ac88f/src/Web.cpp#L1415)

tueddy, 2024-03-04:
The /explorer JSON can be very large. The omission of `dir=false` is intended to save memory in payload transmission.
You are right, the documentation is wrong here.

tueddy, 2024-03-28:
Main issue seems to be fixed, will close it here.  Feel free to reopen or improve with a PR..

closed 2024-03-28 by tueddy